xoseperez/espurna #875: Support for Intertechno V3 protocol?

Thomas-8-Bit, 2018-05-20:
Hi,

i wonder if there will be support for Intertechno V3 protocol in near future. I've got some Intertechno PIR-1000 and Intertechno sockets which use this protocol and can not be used at this moment. I'm not very deep in programming and develop code enhancements and don't know, if it would be possible anyway?

xoseperez, 2018-05-21:
I guess you mean if they can be used with the RF Bridge? I see it's 433,92 and that's good but I don't have the chance to test it. Do you have any specific info on the protocol used?

wildwiz, 2018-05-21:
Maybe the address section actually is a ternary code, like this one http://www.enide.net/webcms/index.php?page=2262-2272-codec ? There are other RCs with 12 symbols address (on/off/float), like this http://www.enide.net/webcms/index.php?page=2262-2272-codec

Thomas-8-Bit, 2018-05-21:
Well i can give you examples:

One of my sockets look like this: 00010010001010101010111111 0 0001
and one of my PIR look like this: 01001111001011001101001010 0 1001

At this moment they work with "Signalduino" but i would love to get them with Espurna.

wildwiz, 2018-05-22:
Just to clear out the obvious items, the Sonoff RF bridge, is not able, just reprogramming the ESP8285 to cover more protocols. That decoding is performed in an auxiliary microcontroller and ESPurna alone cannot work around this.

There are two routes to go beyond this limitation:
* Reprogram the microcontroller with an additional firmware
  https://github.com/Portisch/RF-Bridge-EFM8BB1
* Bypass the microcontroller in hardware
  https://github.com/xoseperez/espurna/wiki/Hardware-Itead-Sonoff-RF-Bridge---Direct-Hack

The first one avoids soldering but the integration with ESPurna is still partial.  
The second one covers anything that the rc-switch library covers.

If you are not afraid by the second route, I have a pending PR that can put the modified bridge in RF trace and extract the timings directly from its radio. You can find it here https://github.com/wildwiz/espurna/tree/on-rx-dump and the documentation is here #854 

If you are able to send me a *complete* (possibly a couple of full frames at least) sequence of pulse timings or a sampled file with the waveform (same as above), I can check if the routines used by the hardware mod can parse it.

Thomas-8-Bit, 2018-05-23:
Hello wildwiz,

i'm not really afraid by soldering but let me know, Espruna is able to work with this hack and will decrypt the protocols right now as far as implemented? So the sockets which work right now will work also after doing the hack?

wildwiz, 2018-05-23:
For completeness, after the hardware mod, you actually *can* switch between bypassed and EFM8BB1 __***receive***__ decoding just switching between the vanilla and the *-direct firmware. 

What you cannot do without touching wires again is switching __***transmit***__ operations, as just one source can drive the transmitter at once (that is why cutting the trace is required).

There is a different way to perform the mod on the TX path that preserves firmware switchability  on TX too, but it is physically way more difficult to perform it (you have not just to cut a thin trace, but solder on it too without making shorts). Given that the modded hardware should cover the features of the original, there is little point in making a more difficult change for gaining nothing (this is different if you are very skilled in hacking SMD boards and want to experiment).

Thomas-8-Bit, 2018-05-24:
Thank you for clarification. I think i choose the easy way. It's ok for me, that TX path is not swtichable, as long as i can transmit via MQTT. I undersstand that this is working but there are other commands to be send via MQTT for transmit without EFM8BB1.

Thomas-8-Bit, 2018-05-25:
So. I've build the hardware hack successfully. I've installed the direct.bin firmware, but it seems, there is no way to get into dump mode. The command quits with OK in telnet session but nothing comes out of it. Also there are no dump-buttons in the debug webif-window. I think, this was the wrong build of die firmware. I cannot find a binary to flash.

I've tried to build from your git-repository but get many errors within atom. Could you please tell me where to get the right firmware binary? Thank you in advance.

wildwiz, 2018-05-25:
Here is a build of the *-direct version with the diagnostic patch applied.
Please remember that is is a dev version, not an official release.
https://raw.githubusercontent.com/wiki/wildwiz/espurna/files/espurna-1.12.6-itead-sonoff-rfbridge-direct-dignostic-ota.bin

Thomas-8-Bit, 2018-05-25:
Thank you very much. The firmware works but there is no output, when an intertechno v3 device is talking. Other devices such sa the ELro AB440 sockes are recognized and there is a debug output. So it seems, that the bridge cannot receive the codes or the frequency of those devices.

wildwiz, 2018-05-25:
Assuming that the frequency is correct, there still is the chance that the code is received but rc-switch libraries don't make it through to the debug point. At this point, I would be curious to see a waveform capture in a wav file or from a scope or digital analyzer :) but I understand that isn't trivial.

Thomas-8-Bit, 2018-05-26:
Well the frequency is 433,92 MHz and i could receive it with a RXb6 receiver module connected to an adruino nano. So it seems, that really the rc-switch libraries don't make it. Maybe there will come a day, that it works. I'm still hoping and waiting.

Thomas-8-Bit, 2018-05-28:
I've found additional information about the protocoll i'm looking for. In this Issue for the EFM8BB1 there is somebody, who catched the protocol with the RF-bridge: https://github.com/Portisch/RF-Bridge-EFM8BB1/issues/10

Is there a way to implement this protocol, even it takes more than 24 bit of data?

wildwiz, 2018-05-30:
Hi, thanks for the heads up. The major problem here is to have reliable pulse timings, as I cannot test the device myself. I will have a look into that post !

closed 2018-08-20 by stale[bot]