xoseperez/espurna #876: Question Emon analog: Calibration value different from read value

mambagr, 2018-05-22:
Hi
I am testing a Wemos D1 mini with an analog current sensor.
I am calibrating using a known load (although not a pure resistive load).
The problem is that after calibration (1893-5W), the value pressented in the status page (>3000W) is waaaay off the calibrated value for the same load.
I don't think it has anything to do with the type of load as it apparenly calibrates and reads the apparent (pun intended) value.
Am i doing something wrong?

xoseperez, 2018-05-23:
The nominal calibration value for the 30A ACS712 is 66mV/A or 15.15A/V. That value is not visible in the web UI but you can log to the device (telnet) and query it:

```
get pwrRatioC
```

mambagr, 2018-05-23:
Using `get pwrRatioC` i get
`$-1`

xoseperez, 2018-05-23:
It should, and the fact that pwrRatioC returns -1 means it's storing the calibration values. Do you click "Save" after entering the values?

mambagr, 2018-05-23:
Yep. I also get the "Settings saved".

mambagr, 2018-05-28:
Verified with another D1 mini. 'get pwrRatioC' returns '$-1'

xoseperez, 2018-06-12:
OK, there was a problem in the EmonSensor. It wasn't recalculating some internal variables based on the stored power ratio, so calibration was not working.

Anyway, I cannot reproduce the issue with ESPurna not storing the pwrRatioC setting.  This is the code responsible for calling the sensor class to recalculate the power ratio and the store it:

https://github.com/xoseperez/espurna/blob/ac7bf70050018157a8eb55c019875fc21229d64d/code/espurna/sensor.ino#L694-L697

As you can see, the only reason to avoid storing the pwrRatioC value is that the pwrExpectedP is 0.

xoseperez, 2018-06-12:
What value do you enter in the "Expected Power" field?

mambagr, 2018-06-18:
Sorry just noticed the email.
I tried with a 1893W heater so i entered "1893W"
I am also troubled that i had to include the "W" symbol or else i could not save (no "Settings saved" popup)

xoseperez, 2018-06-19:
No, it expects a number, no "W" at all.

xoseperez, 2018-06-22:
1.13.0 has been released with changes in this issue. Feel free to reopen if problem persists.

closed 2018-06-22 by xoseperez

HFsi, 2019-06-05:
I'm about to start playing with current measurements.... which version should I try? I have an ESP01, an ACS712 and an ADS1115