xoseperez/espurna #877: Suggestion - Beok TGT70WIFI-EP Smart Wifi Thermostat

MariusLaurinaitis, 2018-05-22:
Hello, what about TGT70 Wi-Fi Series Thermostat ?
[http://www.beok-controls.com/pro_view.asp?id=61](url)
If it uses ESP8266, can ESPURNA support it?

Best regards
Marius

xoseperez, 2018-05-23:
"If" is used an ESP8266 it could be possible to support it, but ESPurna lacks some features to do so:

- NTC support (easy to implement)
- Display
- Configuration (store minimum/maximum temperature for actions)
- Self contained logic (trigger the load when the temp is below X)

So, it is doable but not trivial...

flashy02, 2018-05-28:
Hi, this enhancement would be nice and should additionaly be useable for other devices too. Especially the two last points would be nice for e.g. sonoff-devices like a modified Sonoff-Touch with SHT31:

- Configuration (store minimum/maximum temperature for actions)

- Self contained logic (trigger the load when the temp (or humidity) is below/above X)

With this, someone can use every supportetd ESP8266 Device with T/H sensor to controll e.g. the heating/dehumidifier without central logic like openhab.

Best regards
Michael

fraschizzato, 2019-01-08:
On the same way can be added support for many thermostat. I have a Beca BHT-6000 (Tuya Device). First step is backup original firmware, but how? Here two photos of the device
[image]
[image]

xoseperez, 2019-01-18:
You will have to wire the pads of the TYWE3S module, see the pinout here: #852. Then there comes the hard work: id all the components it uses and what GPIOs are they connected to and what protocols they use...

fraschizzato, 2019-01-18:
Ok, now I'm waiting a second device that i've bougth for tests (the other is already wall mounted and at work), and the ftdi-usb adapter.
When i've wired the pads, is it possible do grab the current firmware and decompile that to get all data?

badigit, 2019-01-18:
> Hello, what about TGT70 Wi-Fi Series Thermostat ?
> [GT70WIFI-EP Smart Wifi Thermostat](http://www.beok-controls.com/pro_view.asp?id=61)
> If it uses ESP8266, can ESPURNA support it?

May be useful: You can control thermostat native via wi-fi with broadlink libraries. It have easy way to integrate with [Homeassistant](https://community.home-assistant.io/t/beta-for-hysen-thermostats-powered-by-broadlink/56267)

fraschizzato, 2019-01-18:
> May be useful: You can control thermostat native via wi-fi with broadlink libraries. It have easy way to integrate with [...]

If is a tuya based device, in homeassistant there is also the tuya component (that works with tuya cloud), is also possible to control the device locally ( https://community.home-assistant.io/t/bht-6000-thermostat-tuya-support/89311 ). That workaround is tested only with bht-6000 thermostat but probably can work with any tuya device.
espurna can be usefull to get better data from the device, example:
- Default firmware does not update target temperature in automatic mode but with custom firmware can be possible
- Default firmware, in local mode, works by request-response logic, but with espurna anything can be pulled-pushed via mqtt

soif, 2019-12-18:
this firmware seems to handle the beca thermostats :
BHT-002-GBLW, BHT-6000, BAC-002-ALW, BHT-002-GCLW 
https://github.com/klausahrenberg/WThermostatBeca

and this one:
Floureon C17.GH3 and MoesHouse BHT-002-GBLW (=Beca ):
https://github.com/bikemike/MikesThermostat

Would be really cool if it could be merged into Espurna. :-)

mcspr, 2019-12-18:
Looking at https://github.com/klausahrenberg/WThermostatBeca/blob/2df063b09bcd5a56616c81802e1ade4b8fc4e5db/WThermostat/WBecaDevice.h#L290 it does look like tuya protocol
https://github.com/xoseperez/espurna/blob/e5b72a0d43892eec7e4496e73031180bd2f3acb7/code/espurna/tuya_transport.h#L87
https://github.com/xoseperez/espurna/blob/e5b72a0d43892eec7e4496e73031180bd2f3acb7/code/espurna/tuya_types.h#L22