xoseperez/espurna #878: new features - support RFID reader

izharm, 2018-05-23:
Hi,

It would be great if you could support an RFID reader, I have good experience with this [one](https://elty.pl/upload/download/RFID/RDM630-Spec.pdf)

supported by this [library](https://playground.arduino.cc/Main/RDM630RFIDReaderLibrary)

thanks,
Izhar

xoseperez, 2018-05-23:
Can't this be used with the UART_MQTT_SUPPORT?

izharm, 2018-05-23:
i thought it could but it doesn't work, there some limited parsing in the library.  Also it may be a termination issue, what's triggering the sending of the UART_MQTT message ?

xoseperez, 2018-05-25:
See 3dae7b71a0390d7e333db2d22aa65a2a62fb0edc

izharm, 2018-05-25:
it works. 
 
changed UART_MQTT_BAUDRATE to 9600
changed  UART_MQTT_TERMINATION to 3

thank you!