xoseperez/espurna #880: influxdb can't accept host parameter

lfarkas, 2018-05-24:
in the influx setup i always got this error message and can't accept any valid parameters:
```
Hostname cannot be empty and may only contain the ASCII letters ('A' through 'Z' and 'a' through 'z'), the digits '0' through '9', and the hyphen ('-')! They can neither start or end with an hyphen.
```
neither:
172.33.22.200
nor:
a-b.xyx.org
so currently it's not possible to configure influxdb.

lfarkas, 2018-05-24:
anyway it should have to be allowed that the hostname be FQDN too which is no not possible since "." are not allowed in the hostname.

wildwiz, 2018-05-24:
about this, let's remember that the underscore is not allowed in host names, but it actually is in domain names, according to
* https://stackoverflow.com/questions/2180465/can-domain-name-subdomains-have-an-underscore-in-it
* https://tools.ietf.org/html/rfc2181#section-11
* https://tools.ietf.org/html/rfc1034

xoseperez, 2018-05-26:
This has been fixed with @mcspr #889 PR

closed 2018-05-26 by xoseperez