xoseperez/espurna #884: Suggestion - Add Xiaomi Smart Desk Lamp

OHertel, 2018-05-24:
Hey there,

maybe somebody has some time to add support for Xiaomi's Smart Desk Lamp? It's running on ESP8266 and there is already an arduino compatible FW: https://github.com/Torxgewinde/Desk-Lamp-Alternative-Firmware

Would be nice, if it could be supported as well. It offers the ability to adjust color temperature and brightness.

Regards
Oliver

xoseperez, 2018-05-26:
Yes, there is one unique feature that should be first implemented before creating a "profile" for it: support for an encoder that would control the brightness...

xoseperez, 2018-08-29:
@OHertel Do you have the chance to test the code in `dev` branch? (build the xiaomi-smart-desk-lamp profile).

closed 2018-09-02 by xoseperez

thehellodonkey, 2019-03-07:
@xoseperez First and foremost, thanks for your excellent work on Espurna. Flashed the current release (1.13.5): Web interface shows correct behavior based on encoder movements (in terms of switch state, warm/cold color balance, and brightness), but the lamp never turns on. Reflashed original firmware afterward, confirming that lamp was not broken during disassembly/soldering.

fluffymadness, 2019-10-05:
Same here. buttons work but the light stays dark. Is there any fix for this?