xoseperez/espurna #886: REST API: one common status page

lfarkas, 2018-05-25:
currently is very complicated to query all sensors data ie.

- first  call /apis
- then call all api endpoint and each call gives only one value.

eg it'd be very useful if i can call a /api/status which gives a "real" json result with all sensor data.

lfarkas, 2018-05-25:
of course it could be a graphql api which would be more user friendly but in this simple case may a rest api would be enough. for the only the query will be important (and wouldn't like to call 3-10 http request just to get 3-10 simple value. i review the current influxdb interface and decided that i'd rather write my own telegraf input plugin. but with this so many call it's rather annoying...
anyway setting all parameters which one can also modify on the web interface would be useful.

thebradleysanders, 2018-06-13:
Enhancement: Add RSSI, Uptime, avg load... To rest api similar to how data is transmitted to MQTT.
This would be a great addition as I want to build a status panel in PHP to see all my Sonoffs without the need of MQTT. Thank you.

Cabalist, 2018-09-11:
RSSI would be great!

mcspr, 2021-10-03:
ref. https://github.com/xoseperez/espurna/pull/2332

closed 2021-10-03 by mcspr