xoseperez/espurna #891: Power down while saving to flash reverts device to factory settings

xoseperez, 2018-05-26:
The emulated EEPROM uses one SPI flash memory sector to store the data. Due to the nature of this flash memory (NOR) an full sector erase must be done prior to write new data. If a power failure (intended or not) happens during this process the sector data is lost and the device will reboot into "factory settings".

Note: writing data to a NOR memory can be done bit by bit (or byte by byte) but only to change a 1 to a 0. The only way to turn 0s to 1s is to perform a sector erase which turns all memory position to 1. But sector erasing must be done in full sectors.

A way to overcome this is to use more than one sector to store data and check on boot which one is the latest valid sector available. This is the way Tasmota works.

The goal would be to wrap the stock EEPROM class so it handles this multiple-sector. The _sector class variable in the EEPROM class is protected, so accessible from a inherited class, an it's state-less. This should allow to create a class that wraps the EEPROM class, reads from the different, predefined, available sectors, chooses the valid one, reads it's data into memory and hop to the next sector after each successful write.

xoseperez, 2018-05-28:
Testing a solution based on the [EEPROM_Rotate](https://github.com/xoseperez/eeprom_rotate) library in the eeprom_rorate branch of this repo.

pabloalcantara, 2018-06-11:
All of my 5 sonoff pow with ESPurna 1.12.6 lost config after a power failure.
Do you have any test version pré compiled with this possible fix?

xoseperez, 2018-06-11:
The eeprom_rotate branch will be relesed this week, and prevents this issue

xoseperez, 2018-06-11:
Merged to the dev branch. You will need version 0.9 of the EEPROM_Rotate library (https://github.com/xoseperez/eeprom_rotate). If you are using PlatformIO it will download it automatically. Otherwise checkout that version from the repo and install it to your IDE.

Staged for release.

closed 2018-06-11 by xoseperez