xoseperez/espurna #895: Can't set new password

88adong88, 2018-05-27:
V1.12.17
new admin password:
Admin
Admin123
Admin123@

The password you have entered is not valid, it must have at least 5 characters, 1 lowercase and 1 uppercase or number!

cent0, 2018-05-27:
If you have build source,you have make one error in the setting file.

Same my error, i need reflash espurna with wire and serial adapter.

Regards

mcspr, 2018-05-27:
*Oops. I did misled some people with mistaking it with hostname issue.*
There's breaking change from https://github.com/xoseperez/espurna/commit/8a02c9fd6941e0e2035eb0fc7abf245ad8c14318#diff-35d45b13e9460cb72e90b29e1c6e56a0L145

`new RegExp('...')` must be escaped differently than `/...`/. idk if this was intended, but it should be like this:
```patch
@@ -142,7 +142,7 @@ function validateForm(form) {
-    var re_password = new RegExp('^(?=.*[A-Z\d])(?=.*[a-z])[\w~!@#$%^&*\(\)<>,.\?;:{}\[\]\\|]{5,}$');
+    var re_password = new RegExp('^(?=.*[A-Z\\d])(?=.*[a-z])[\\w~!@#$%^&*\\(\\)<>,.\\?;:{}\\[\\]\\\\|]{5,}$');
```

cent0, 2018-05-27:
hello i hare rebuild with this change in custom.js file,  
-    var re_password = new RegExp('^(?=.*[A-Z\d])(?=.*[a-z])[\w~!@#$%^&*\(\)<>,.\?;:{}\[\]\\|]{5,}$');
+    var re_password = new RegExp('^(?=.*[A-Z\\d])(?=.*[a-z])[\\w~!@#$%^&*\\(\\)<>,.\\?;:{}\\[\\]\\\\|]{5,}$');

@cent0

Attach files by dragging & dropping,

, or pasting from the clipboard.
Styling with Markdown is supported

and same problem not working new password

i have build espurna-dev.zip  code  for sonoff pow interface

have changed platformio.ini 
this line 
env_default = itead-sonoff-pow

in sensor.h 
this line 
#define SENSOR_POWER_CHECK_STATUS           0

login work ok and password pdate have this problem
The password you have entered is not valid, it must have at least 5 characters, 1 lowercase and 1 uppercase or number!

regards

cent0, 2018-05-27:
> There's breaking change from 8a02c9f
> #diff-35d45b13e9460cb72e90b29e1c6e56a0L145

My constatation is this,

If reprogramming by wire and not have erased successfully the flash, not
working the change password,
I have test and and problem is not fixed.

I have resolved in this mode:

1-Reprogramm  with  1.12.4  set new password.
2-Update 1.12.4 by web in the admin menu, with my building version 1.12.7a
with source dev .

Best regards.

Your other have found this problem ?

mcspr, 2018-05-27:
@cent0 Maybe your email client does something to the formatting? Check the message in the web, there is link in the email's footer.
https://jsfiddle.net/ry6qvqux/ - testing regexp, it works

And you could've used serial connection to set it, no need to re-flash :) `set adminPass Admin123`

cent0, 2018-05-27:
Tested Admin1234@  Not working
And tested more password
Espurna12
Espurna12@
 not apply and not working 😠😠

I have like your post and script

A this time i have solved and bypass the problem with my pleasure and i
have this Password 😀
Admin123

Programm with old version change password
Reprogramm with 1.12.7a
And have Fixed password,

If you reset interface is impossibile save new password,
You have login with fibonacci
And new password is not settable.😠😠😈

mcspr, 2018-05-27:
https://github.com/xoseperez/espurna/wiki/WebInterface#web-interface-build-process
just changing index.html would not suffice, you need to run gulp too.

cent0, 2018-05-27:
I have build with atom Platformio. And not work

I have wrong setting, and have compiled with error?
Do you have guide for compiling setting?

CrashDrives, 2018-05-27:
I have contourned the problem by flashing an older version of espurna and upgraded in the web interface with the last version.

If that can help you

xoseperez, 2018-05-29:
This was fixed with ad623524f515ee6157eb3277d54a0a83e357a7cb. Staged for release.

xoseperez, 2018-05-29:
Also, see #879

closed 2018-05-29 by xoseperez