xoseperez/espurna #896: Backup settings are truncated

meyergru, 2018-05-27:
Whenever I backup my settings, the resulting JSON file is truncated at some arbitrary point, mostly around 1500 bytes.

This results in a corrupted JSON that cannot be used to restore the settings afterwards.

xoseperez, 2018-05-28:
The max size of the file might be limited by the available memory since right it created before being sent. Can you check how much free heap you have before trying to do the backu? You can use `heap` command from the debug window.

mcspr, 2018-05-28:
@xoseperez I did test this yesterday and noticed that content-length is wrong - it is for the not-yet-pretty json. `prettyPrintTo` returns real size that should be used.
Tested with fresh device, after saving once from web. Around 14k heap backup was successful - 2313 bytes, not cut randomly. With lower heap (12952) it did not even start file download.

xoseperez, 2018-05-28:
I guess the best option would be to "stream" the contents as they are generated...

meyergru, 2018-05-28:
I could not look at the heap size before now, since apparently, the web UI is not reachable via a proxied connection?

The free heap is 17904 bytes when the output gets truncated.

mcspr, 2018-06-11:
Amending my previous reply... I did some tests since, and have some things to note:
- AsyncResponseStream already does keep printed json string in `cbuf` object. Changing it to chunked response does not make much difference.
- Removing `webLog()` from the top of config method avoids truncation. Debug code sends websocket message and immediately begins sending large config, which somehow causes tcp stack to run out of memory (?)
This is the most direct workaround for this.
- There is similar truncation when running `keys` in telnet and web. Adding nop delay by calling `optimistic_yield(100)` at the end of `debugSend()` in case it has TELNET or WEB sessions active allows all data to pass.

And some calls to DEBUG_MSG can be optimized to happen less frequently *(plus, they have some interesting logic for timestamping...)*

xoseperez, 2018-06-11:
Interesting... @mcspr do you mind issuing a PR with your suggested changes?
What do you mean by "interesting logic"?

mcspr, 2018-06-11:
@xoseperez Sure. Just need to check if weblog can still work.

*// about DEBUG_MSG*
It allows line continuation and requires remembering how data is presented from the code before. Accidentally, while trying to reduce debugsend calls, I did replace much of these with string formatting.
https://github.com/xoseperez/espurna/blob/f1c2aa1b819fc159de530cf9194dc067c478be22/code/espurna/ota.ino#L238-L239
https://github.com/xoseperez/espurna/blob/f1c2aa1b819fc159de530cf9194dc067c478be22/code/espurna/utils.ino#L268-L275

e.g. for modules listing:
```patch
-    DEBUG_MSG_P(PSTR("[INIT] SUPPORT:"));
+    DEBUG_MSG_P(PSTR("[INIT] SUPPORT:%s\n"), getEspurnaModules().c_str());
```

```cpp
PROGMEM const char espurna_modules[] =
    #if ALEXA_SUPPORT
        " ALEXA"
    #endif
    #if BROKER_SUPPORT
        " BROKER"
    #endif
...
        "";

String getEspurnaModules() {
    return String(FPSTR(espurna_modules));
}
```

xoseperez, 2018-06-12:
(Sorry, that commit should not be linked to this issue, my bad)

xoseperez, 2018-06-13:
#930 has been merged to dev. I have tested it and looks good. @meyergru, can you validate this fixes the issue.

meyergru, 2018-06-13:
I have no development environment to compile dev... if anybody can provide the binary for the Sonoff bridge, I can test it. Otherwise, it will have to wait until the weekend.

mcspr, 2018-06-14:
@meyergru here's recent git version via travis, built like releases here - https://github.com/mcspr/espurna-travis-test/releases/tag/1.13.0c

@xoseperez Would it make sense to delay sending webLog data until the next loop? I got a feeling some of the problems are from doing network request from inside async callback. Or from other log call in some other callback.

mcspr, 2018-06-14:
Damn :/ Only other theory I have is EEPROM somehow breaks response if called to frequently. Which is the case for both `/config` and `keys`

xoseperez, 2018-06-21:
Dumping the settings line by lines works fine, the drawback is that they are not alphabetically sorted...

mcspr, 2018-06-21:
For web another solution is to just generate file blob from the form, with pure js. No TCP connections that way.

I'll push later today example demonstrating this

xoseperez, 2018-06-22:
Released with 1.13.0. Closing. @meyergru and @mcspr , feel free to open again if problem persists. Or open a new issue/PR for a different approach.

closed 2018-06-22 by xoseperez

meyergru, 2018-06-22:
I just tried version 1.13.0, but the file is still corrupted and cannot be imported.
I think this is mainly because of the missing closing bracket. Maybe that was forgotten in the new output scheme?

The file has 125 lines and is ~3000 bytes now, so it seems as if it is not truncated other than the missing bracket.

meyergru, 2018-06-23:
Used the wrong build, all is well with 1.13.0.