xoseperez/espurna #897: Sonoff Dual R2 not responding after successful flashing

skyynet, 2018-05-27:
I successfully flashed one Electrodragon and some Sonoff Dual R1 boards with ESPurna 1.12.6.
I also flashed two R2 boards with 1.12.6 and of course changed the configuration in arduino.h to the r2 setting.
Unfortunately both boards do not blink or show a wifi ap after having them flashed. When the first board failed I thought I might have killed it but the second one also behaves the same.

Any idea?

cent0, 2018-05-27:
Board not reply by serial connection ?
Erase flash

skyynet, 2018-05-27:
I thought the Sonoff Dual boards don't support serial monitor.
I compiled with option serial and debug level core but don't get any output. Am  I supposed to enter something in the serial monitor?
I also flashed with option ERASE FLASH All Flash Contents but nothing changes.
I can flash the board without any problem but it doesn't respond afterwards.
I thought maybe there's some issue in the firmware for Dual R2 boards.

xoseperez, 2018-05-29:
The dual uses the UART to communicate with the secondary microcontroller, so you don't have serial debug.
I have not tested a dual with the latest versions and I don't have one now to do so... I will have to try later this week. Maybe you can try flashing a simple sketch (a blink) to see if the board is ok...

skyynet, 2018-06-07:
I thought the R2 doesn't use a secondary microcontroller anymore? Should I send you one Sonoff Dual R2 so you can check it on your own?

xoseperez, 2018-06-11:
Sorry, you are right. Then you should be able to get some info through serial if the flashing went ok.

xoseperez, 2018-06-24:
If you are not getting a single line of boot log I would say your device is broken. I would double check that you are using 115200 baud communication and that the device is working flashing a simple sketch that blinks GPIO13.

skyynet, 2019-06-08:
I was able to flash 1.13.5 today via uart. Don't know what caused the problems in the past. Closing.

closed 2019-06-08 by skyynet