xoseperez/espurna #899: Wifi settings gone after changing the MQTT server

jsponz, 2018-05-28:
With the last version, after changing the MQTT server (just the URL), the Wifi settings are gone. I need to configure Wifi again. 

Any hint?

xoseperez, 2018-05-29:
Only the WiFi settings? All the other settings remained?

jsponz, 2018-05-29:
Yes, but after the compilation, I just add the WiFi details. I don't change anything else. 

I wanted to use another MQTT broker and after clicking on Save and Reconnect, I lost the connection and the ESP AP appeared...

xoseperez, 2018-05-29:
Can you reproduce the issue? Does it always happen?

jsponz, 2018-05-30:
If I disable MQTT, click on Save, change MQTT broker, click on Save, and enable MQTT again, it works. 

It seems that changing MQTT broker while connected, disturbs the code :-)

xoseperez, 2018-06-04:
I cannot reproduce the issue. Changing the broker just disconnects and reconnects to the new broker, it keeps the WiFi connected...

Can you use a serial monitor to get the logs of the device while this event is happening?

closed 2018-08-20 by stale[bot]