xoseperez/espurna #901: Sonoff POW R2 crashes after changing initial admin password

ruimarinho, 2018-05-29:
After the initial flash (using latest commit https://github.com/xoseperez/espurna/commit/36f6d399bd5abba4e54bfae182f0f952fffae61d from dev branch), the Sonoff POW R2 boots correctly with the default ESPURNA SSID.

As usual, I enter the default password to connect to it and then navigate to the device's IP where I'm presented with a password change page. After changing the default password, the device disconnects and the wifi responds to correctly to the new password but the web server does not start — the device always enter safe mode.

My device has board 2017-11-30 V1.0 with CSE77598-SG (1741D1A).

xoseperez, 2018-06-04:
I cannot reproduce the issue. How do you know the device enters safe mode? Can you still connect to it via AP? Can you type `crash` to dump the latest crash stack?
BTW, the latest commit changes the way the authentication works, it could affect this issue too.

ruimarinho, 2018-06-05:
The web server does not start but the telnet server is. The console output says the device is in safe mode.

Regarding the `crash` command, please see the output at https://github.com/xoseperez/espurna/issues/761#issuecomment-388594434

ruimarinho, 2018-06-17:
@xoseperez is there any way I can help debug this issue?

ServerCare, 2018-06-19:
Just wanted to add that I seem to have the same or similar issue. Possibly my description of the issue may assist. I actually have this now happening on three new R2's that were flashed with the stable build. In my case after changing name things seem to be fine still, but after I try to add them to my wifi (using a fixed IP) things go wrong. From that point on the devices can only be seen as wifi AP's (curiously with the changed name in two devices). Connecting to the AP works, I can ping to 192.168.4.1 but web- browsing to it does not work. In addition: all three devices report ' -ERROR buffer overflow' when you telnet to them.

What can I do to 'reset' the devices?

ServerCare, 2018-06-19:
Did some further testing with a 4rd R2 (yes I have plenty of them ;-) 
This time I was more careful with the setup procedure and I disabled 'scan networks' (didn't work anyway) did not change the name and left the IP up to DHCP.

This time I can connect & things seem to work, however; very soon after bootinf the device no longer accepts commands in telnet or debug window, simply responding -error unknown command (multiple times, at random) and error- buffer overflow (in the telnet window). The debug window typicall responds with '[WEBSOCKET] Requested action: dbgcmd' when I type in (for example) 'info'

mcspr, 2018-06-19:
@ServerCare Would you mind trying this build?
https://github.com/mcspr/espurna-travis-test/releases/download/20180619/espurna-1.13.0c.gitb7915c3-itead-sonoff-pow-r2.bin
https://github.com/mcspr/espurna-travis-test/releases/tag/20180619

By the looks of it serial connection is active. And that was solved by this commit in the `dev`: https://github.com/xoseperez/espurna/commit/40b9fdf3b4b733a1baae6368a3c50f2b72fcccf7

ServerCare, 2018-06-19:
I have done an OTA update on the one device that I could still access over the web interface (which worked like a charm!) and can confirm that the telnet issues are now over. Output looks fine after the update.

Would this also fix the issues I have on my other 3 devices that I cannot reach anymore? I'll see if I can update them using the flash procedure tomorrow.

Some more good news; after the upgrade I have changed the name of the device as well as changed to a static IP without any issues. AND I was able to scan the network without any issues...

ServerCare, 2018-06-20:
Further good news: I have re-flashed my three unresponsive R2's, all have come back to life & are now acting as expected. Thanks!

@xoseperez just to let you know this fix is of a huge impact on all R2's I have implemented this on (4 in total) In essense: without this fix the current stable is not reliable for the R2.

xoseperez, 2018-06-20:
@ruimarinho Can you confirm if @ServerCare is having the same issue and if de branch is working for you?

ruimarinho, 2018-06-21:
Will try to erase flash and reflash with master this weekend.

I have tried multiple times to run master after I had confirmed 1.12.6 was
unstable for me, but never got to a working point. Perhaps I was simply
unlucky with the commits I was testing with :)

ruimarinho, 2018-06-25:
@xoseperez tried with 1.13.1a (before I also erased flash using esptool.py) but still not working. Now I cannot even get an IP via DHCP. Using a static IP of 192.168.4.1, I was able to telnet into the device (the web server was down) and also connected to serial.

```
INFO
[189695]

[189695] [INIT] ESPURNA 1.13.1a
[189696] [INIT] xose.perez@gmail.com
[189696] [INIT] http://tinkerman.cat

[189696] [INIT] CPU chip ID: 0x4EB844
[189697] [INIT] CPU frequency: 80 MHz
[189697] [INIT] SDK version: 1.5.3(aec24ac9)
[189697] [INIT] Core version: 2.3.0
[189698] [INIT] Core revision:
[189698]
[189698] [INIT] Flash chip ID: 0x1640EF
[189700] [INIT] Flash speed: 40000000 Hz
[189700] [INIT] Flash mode: QIO
[189700]
[189700] [INIT] Flash size (CHIP)   :  4194304 bytes / 1024 sectors (   0 to 1023)
[189702] [INIT] Flash size (SDK)    :  1048576 bytes /  256 sectors (   0 to  255)
[189702] [INIT] Reserved            :     4096 bytes /    1 sectors (   0 to    0)
[189703] [INIT] Firmware size       :   476928 bytes /  117 sectors (   1 to  117)
[189703] [INIT] Max OTA size        :   544768 bytes /  133 sectors ( 118 to  250)
[189704] [INIT] EEPROM size         :     4096 bytes /    1 sectors ( 251 to  251)
[189704] [INIT] Reserved            :    16384 bytes /    4 sectors ( 252 to  255)
[189705]
[189705] [INIT] EEPROM sectors: 251, 250
[189705]
[189706] [INIT] BOARD: ITEAD_SONOFF_POW_R2
[189707] [INIT] SUPPORT: ALEXA BROKER DEBUG_TELNET DEBUG_WEB DOMOTICZ HOMEASSISTANT MDNS_SERVER MQTT NTP SCHEDULER SENSOR TELNET TERMINAL THINGSPEAK WEB
[189707] [INIT] SENSORS: CSE7766
[189707]
[189708] [INIT] Last reset reason: Power on
[189708] [INIT] Settings size: 266 bytes
[189709] [INIT] Free heap: 13864 bytes
[189710] [INIT] Power: 3007 mV
[189710] [INIT] Power saving delay value: 10 ms
[189710]
[189710] [WIFI] ------------------------------------- MODE STA
[189710] [WIFI] SSID
[189710] [WIFI] IP    0.0.0.0
[189711] [WIFI] MAC   2C:3A:E8:4E:B8:44
[189711] [WIFI] GW    0.0.0.0
[189711] [WIFI] DNS   208.67.222.222
[189711] [WIFI] MASK  0.0.0.0
[189711] [WIFI] HOST  http://ESP_4EB844.local
[189711] [WIFI] BSSID FF:FF:FF:FF:FF:FF
[189712] [WIFI] CH    1
[189712] [WIFI] RSSI  31
[189712] [WIFI] -------------------------------------- MODE AP
[189712] [WIFI] SSID  ESPURNA-4EB844
[189712] [WIFI] PASS  fibonacci
[189712] [WIFI] IP    192.168.4.1
[189713] [WIFI] MAC   2E:3A:E8:4E:B8:44
[189713] [WIFI] ----------------------------------------------
[189713] +OK
```

```
KEYS
[139737] Current settings:
[139738] > board => "71"
[139738] > boardName => "ITEAD_SONOFF_POW_R2"
[139738] > btnGPIO0 => "0"
[139739] > btnRelay0 => "0"
[139739] > cf1GPIO => "13"
[139739] > cfGPIO => "14"
[139739] > cfg => "3"
[139739] > hostname => "ESPURNA-4EB844"
[139740] > ledGPIO0 => "15"
[139740] > ledLogic0 => "1"
[139740] > mqttEnabled => "0"
[139741] > relayBoot0 => "0"
[139741] > relayGPIO0 => "12"
[139741] > relayPulse0 => "0"
[139741] > relayTime0 => "0.00"
[139741] > relayType0 => "0"
[139741] > selGPIO => "5"
[139742] Number of keys: 17
[139742] Current EEPROM sector: 250
[139743] Free EEPROM: 3830 bytes (93%)
[139743] +OK
```

```
CRASH
[210885] [DEBUG] No crash info
[210927] +OK
UPTIME
[222667] Uptime: 222 seconds
[222668] +OK
ERASE.CONFIG
```

Then reconnecting via telnet shows:

```
CRASH
[076058] [DEBUG] Latest crash was at 226863 ms after boot
[076059] [DEBUG] Reason of restart: 2
[076059] [DEBUG] Exception cause: 29
[076059] [DEBUG] epc1=0x4020cf07 epc2=0x00000000 epc3=0x00000000
[076059] [DEBUG] excvaddr=0x00000000 depc=0x00000000
[076060] [DEBUG] >>>stack>>>
[DEBUG] 3fff1f80: 00000000 00000001 ffffffe6 4020cf03
[DEBUG] 3fff1f90: 40104ac0 0432c3cf 3fff2060 4021b895
[DEBUG] 3fff1fa0: 3fff07f0 3fff2060 3fff0ff0 3fff2060
[DEBUG] 3fff1fb0: 3fffdad0 00000001 3fff07f0 4021c39d
[DEBUG] 3fff1fc0: 402017e6 0000000a 0000000a 3fff0fe4
[DEBUG] 3fff1fd0: 3fffdad0 3fff0bd8 3fff07ec 40202bef
[DEBUG] 3fff1fe0: 00000000 3fff0bd8 00000002 40204120
[DEBUG] 3fff1ff0: 3fffdad0 00000000 3fff0fde 40229c08
[DEBUG] 3fff2000: feefeffe feefeffe 3fff0ff0 401006fc
[DEBUG] <<<stack<<<
[076106] +OK
```

and on serial monitor just:

```
 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
```

xoseperez, 2018-06-25:
I see that the board is using QIO flash mode, but it should be DOUT....

[189711] [WIFI] MASK  0.0.0.0

> [189711] [WIFI] HOST  http://ESP_4EB844.local
> [189711] [WIFI] BSSID FF:FF:FF:FF:FF:FF
> [189712] [WIFI] CH    1
> [189712] [WIFI] RSSI  31
> [189712] [WIFI] -------------------------------------- MODE AP [...]
-- 

Xose Pérez
@xoseperez
http://linkedin.com/in/xoseperez
http://tinkerman.cat

ruimarinho, 2018-06-25:
@xoseperez I am using platformio v3.5.2. Any idea?

xoseperez, 2018-06-25:
I see... the 3.5.3 version of PlatformIO deprecated the `flash_mode` tag but if you are using an older version it will default to QIO... Upgrade your PlatformIO Core to 3.5.3 and rebuild.

ruimarinho, 2018-06-25:
@xoseperez that was indeed the case for that particular issue. I wonder how a breaking change like this would land in a patch release..

However, even in DOUT mode, a crash is still persistent:

```
INFO
[063008]

[063009] [INIT] ESPURNA 1.13.1a
[063009] [INIT] xose.perez@gmail.com
[063009] [INIT] http://tinkerman.cat

[063009] [INIT] CPU chip ID: 0x4EB844
[063010] [INIT] CPU frequency: 80 MHz
[063010] [INIT] SDK version: 1.5.3(aec24ac9)
[063011] [INIT] Core version: 2.3.0
[063011] [INIT] Core revision:
[063012]
[063012] [INIT] Flash chip ID: 0x1640EF
[063015] [INIT] Flash speed: 40000000 Hz
[063015] [INIT] Flash mode: DOUT
[063016]
[063016] [INIT] Flash size (CHIP)   :  4194304 bytes / 1024 sectors (   0 to 1023)
[063019] [INIT] Flash size (SDK)    :  1048576 bytes /  256 sectors (   0 to  255)
[063020] [INIT] Reserved            :     4096 bytes /    1 sectors (   0 to    0)
[063020] [INIT] Firmware size       :   476928 bytes /  117 sectors (   1 to  117)
[063021] [INIT] Max OTA size        :   544768 bytes /  133 sectors ( 118 to  250)
[063021] [INIT] EEPROM size         :     4096 bytes /    1 sectors ( 251 to  251)
[063022] [INIT] Reserved            :    16384 bytes /    4 sectors ( 252 to  255)
[063022]
[063023] [INIT] EEPROM sectors: 251, 250
[063023]
[063023] [INIT] BOARD: ITEAD_SONOFF_POW_R2
[063025] [INIT] SUPPORT: ALEXA BROKER DEBUG_TELNET DEBUG_WEB DOMOTICZ HOMEASSISTANT MDNS_SERVER MQTT NTP SCHEDULER SENSOR TELNET TERMINAL THINGSPEAK WEB
[063025] [INIT] SENSORS: CSE7766
[063025]
[063026] [INIT] Last reset reason: Power on
[063026] [INIT] Settings size: 266 bytes
[063027] [INIT] Free heap: 15824 bytes
[063029] [INIT] Power: 2951 mV
[063030] [INIT] Power saving delay value: 10 ms
[063030]
[063030] [WIFI] -------------------------------------- MODE AP
[063031] [WIFI] SSID  ESPURNA-4EB844
[063032] [WIFI] PASS  fibonacci
[063032] [WIFI] IP    192.168.4.1
[063032] [WIFI] MAC   2E:3A:E8:4E:B8:44
[063033] [WIFI] ----------------------------------------------
[063033] +OK

CRASH
[070819] [DEBUG] Latest crash was at 2074 ms after boot
[070820] [DEBUG] Reason of restart: 2
[070820] [DEBUG] Exception cause: 28
[070820] [DEBUG] epc1=0x40204118 epc2=0x00000000 epc3=0x00000000
[070820] [DEBUG] excvaddr=0x00000000 depc=0x00000000
[070820] [DEBUG] >>>stack>>>
[DEBUG] 3fff1fe0: 3fffdad0 3fff0bd8 0000000a 40204120
[DEBUG] 3fff1ff0: 3fffdad0 00000000 3fff0fde 40229c08
[DEBUG] 3fff2000: feefeffe feefeffe 3fff0ff0 401006fc
[DEBUG] <<<stack<<<
[070863] +OK
```

xoseperez, 2018-06-26:
`crash` will report the latest crash info, it doesn't mean that crash is recent. You log shows that you just powered on the board. You might want to wait and check the log using `info` until the "Last reset reason" shows "Exception" and then type `crash` to get the stack for that exception.

What I see in your previous message when typing `keys` is that it doesn't have any network defined so it will hardly connect to your home router... did you do  a factory reset maybe?

ruimarinho, 2018-06-26:
Yes, in addition to erase_flash using esptool. I can’t get the device to
provide a DHCP address in AP mode. No web server as well.

What could I be missing?

ruimarinho, 2018-07-09:
@xoseperez any additional feedback I can provide?

xoseperez, 2018-07-10:
Let's recap.
The device says it's in AP mode (the SSID should be ESPURNA-4EB844) but you cannot even connect to it. Is that true?

ruimarinho, 2018-07-11:
@xoseperez yes. The SSID is exactly like that but it doesn't provide the client a dynamic IP. If I manually set one and its gateway, I am able to connect via telnet, but nothing else. I can never see the web interface.

I am running `PlatformIO, version 3.6.0a3`, using the latest dev code and flashing after running `esptool.py --port /dev/tty.usbserial-A50285BI erase_flash` to make sure no artefacts are left from previous builds.

At this point, I have either a) received a bad unit, b) my unit includes some modifications which makes it incompatible with the current code or c) am using esptool incorrectly. Note that I have flashed several working non-R2 devices.

ruimarinho, 2018-07-11:
@xoseperez the pull from `87859c5a..c16b3800` appears to have fixed the issue! As usual, I've erased the flash before trying out the new build.

I've noticed a few bugs like being asked twice for the access password on Safari or even the initial password update not providing any UI feedback (even though it's working), but nothing compared to the initial scenario.

Any idea which commit(s) fixed this?

xoseperez, 2018-07-12:
There has been no change related to the POW since 87859c5... maybe the issue is related to something else...

xoseperez, 2018-07-16:
You debug log says you are using SDK 2.3.0, the default for ESPurna although 2.4.1 is already available.

xoseperez, 2018-08-04:
No. I meant just that. It can't be a platform update since you are still in 2.3.0, like it should be.
The only way to really find the reason is to downgrade to somewhere before 87859c5 to see if the problem is still there.