xoseperez/espurna #905: Recommended platform to build binaries with command line

fpalinf, 2018-05-30:
Is there a recommended platform to build binaries with command line ?
I suspect running code/build.sh is the most straight forward process ? Is using a python/2.7 docker image would make the things flawless ?

wildwiz, 2018-05-30:
code/build.sh itead-sonoff-basic
the available builds are defined code/platformio.ini, as [env:itead-sonoff-basic]

xoseperez, 2018-06-04:
The `code/build.sh` script calls platformio to build the image, renames the result binary and moves it to a `firmware` folder. You can use it the way @wildwiz says.
You can also use platformio directly, and flash the image to the device.
But I think your question goes beyond how to build the image. The bash script is an only-linux tool, of course (maybe it would work on mac too). But platformio is python and cross-platform, so it should work on any OS and platform as long as it supports python. I have even used a PocketChip to build and flash "on the field" :smile:

closed 2018-06-04 by xoseperez