xoseperez/espurna #908: Sonoff pow adding function. IF ACTIVE POWER > 10W* switch on for 6 sencod* , *=adjustable.

cent0, 2018-05-31:
hello it is possible to add the following function to bypass the excess energy, if active energy is lower than a certain value that can be set manually, then switch on for certain seconds. if active energy is greater than the set value switch off.

This function is more appreciate for bypass Mqtt and domoticz.and one raspberry

😀
This is one other idea for extra function
In case it would also be possible to send a switch on or a switch off addressed to another settable id

xoseperez, 2018-06-04:
Pretty specific. I would do that logic somewhere else. Not hard to implement with Node-RED, for instance...

closed 2018-06-04 by xoseperez

cent0, 2018-06-05:
since to send to other devices you are advised to use node-red. it would be at least possible to insert the relay activation function if the active current is <10W* ?

Or help for find code to mod and compile. Without graphic html
Best regards.

cent0, 2020-05-31:
> Pretty specific. I would do that logic somewhere else. Not hard to implement with Node-RED, for instance...

Hi nodered need one raspberry or one other cpu for database and more...

I need only read the MAGNITUDE_POWER_ACTIVE
And add one istance ...if magnitude ....>?....
     
SEND ONE STRING  HTTP.BEGIN

Thank you