xoseperez/espurna #912: Reboot and factory reset via MQTT

jsponz, 2018-06-02:
Is it possible to make a reboot and a factory reset via MQTT? 

If so, what should be the command?

Thanks!

wildwiz, 2018-06-02:
my 5 cents ... I feel this a factory reset a bit too 'hard' to be requested via MQTT, especially as after honoring the command you would be cut off from the device (the mqtt server to contact would be wiped out by the command along with MQTT enabling as well). 

However might make sense to request just a 'plain' reset where. supposing that current configuration has MQTT already, after the reset you would regain the link and be able to control it again.

jsponz, 2018-06-02:
Thanks for the contribution. "reboot" is already included in the code, but I want to know how to invoke it. 

Regarding "factory reset", it would be useful to get to factory values, considering you define the MQTT broker as part of default values.

mcspr, 2018-06-03:
@jsponz is right and there is already code that does reboot through MQTT (called through "actions" topic, "reboot" is the only valid payload)
https://github.com/xoseperez/espurna/blob/dff81d6795c4b6647463645c2f484be578d10562/code/espurna/mqtt.ino#L375-L378
https://github.com/xoseperez/espurna/blob/53aa1d109e41cd8ec06291ba2747157be128450b/code/espurna/config/general.h#L699
`$ mosquitto_pub ... -t <root>/action/set -m reboot` does the reboot

Another is unfinished HTTP Api handler with GET(?!) request:
https://github.com/xoseperez/espurna/blob/dff81d6795c4b6647463645c2f484be578d10562/code/espurna/api.ino#L159
`$ curl -v 'http://admin:fibonacci@192.168.1.5/rpc?apikey=...&action=reboot`

BTW HTTP config api can be used to send whole config, without dependence on build time defines:
https://github.com/xoseperez/espurna/issues/751#issuecomment-378590742

jsponz, 2018-06-03:
@mcspr Thanks for the hint. It works!

My idea of the factory reset via MQTT is for this case:

- I have a version X and I have made some changes/configurations
- I update the node via update server (nofuss) with version Y
- I want to make a factory reset via MQTT so the new settings of version Y take place.

xoseperez, 2018-06-04:
Reset to factory settings will remove everything, including WiFi and MQTT configuration, the device won't connect to your network and of course will not update with new versions... I'm not sure this makes sense... but it's certainly something that won't be implemented via MQTT.

closed 2018-06-04 by xoseperez

pniaps, 2018-06-05:
This is not totally true, it makes sense when we set a hardcoded wifi and mqtt configuration, although it is true that for most use cases it may not make sense

https://github.com/xoseperez/espurna/blob/53aa1d109e41cd8ec06291ba2747157be128450b/code/espurna/config/general.h#L293

https://github.com/xoseperez/espurna/blob/53aa1d109e41cd8ec06291ba2747157be128450b/code/espurna/config/general.h#L297

https://github.com/xoseperez/espurna/blob/53aa1d109e41cd8ec06291ba2747157be128450b/code/espurna/config/general.h#L569

sfilinto, 2020-09-08:
> This is not totally true, it makes sense when we set a hardcoded wifi and mqtt configuration, although it is true that for most use cases it may not make sense [...]

i agree with your statement. i hard code wifi with other details & distribute using nofuss. need a simple way to "factory reset" the device after the new firmware is updated.  i hope this request will be re-considered. @xoseperez