xoseperez/espurna #914: H801 Wifi Connection problems with Firmware 1.12.5 and 1.12.6

bigd-73, 2018-06-03:
Hallo, 

Firmware 1.12.5 and 1.12.6 have massive problems with the Wifi connection. A connection is only established if I disconnect the power supply briefly. After a few minutes, the connection is lost again. a reboot does not succeed. The device builds up its own hotspot until i power off the device. Version 1.12.4 works fine.

Regards Mario

flashy02, 2018-06-05:
Hi, is this really a problem of the Version 1.12.6? I'm using this version with my H801 without any WiFi-Problems.

Regards Michael

bigd-73, 2018-06-05:
hi, i can only say that when i run the version 1.12.4 on the same device i don't have any problems. 
with the version 1.12.6 i have the mentioned problems. i use the pre-combined versions from the gitbub site. 

regards mario

mrcodegh, 2018-06-05:
I'm seeing OK WiFi with 1.12.6.

I wonder how well the ESP8266 power supply is regulated in the H801 relative to various LED loads.  I've seen weird WiFi when 3.3v is not stable on other projects using the ESP8266.

@bigd-73, have you tried running with reduced or no load?  This may not explain working with 1.12.4 unless maybe it controls the load differently.  I don't know, just guessing here.

Regards, Mike.

CH-Tefi, 2018-08-05:
If have the same probleme with some H801, stable connection until version 1.12.4.
After that, including 1.13.1 after some second no wifi connection.