xoseperez/espurna #915: ACmedia remotes not recognized by sonoff-rf-bridge-direct

wildwiz, 2018-06-04:
> @derFrickler wrote in followup to #854 and #864: Just tested the diagnostic image. With my working  intertechno remotes i receive a lot of output lines with Trace RF  enabled, ending up with sending the received code to the mqtt broker. With the non-working remote this is all i receive on one keypress: [096213] [RFBRIDGE] RF buffer count: 84, data: 5600 2474 1621 312  274 708 674 316 672 314 669 316 270 715 276 707 673 315 273 709 675 311  274 709 278 708 676 311 274 709 676 313 276 703 678 312 673 312 279 706  279 704 282 705 282 700 283 703 678 307 678 307 681 307 280 702 282 705  282 702 285 No newline after receiving the code. A 2nd keypress on the remote just appends to this line. so like this: [322665] [RFBRIDGE] RF buffer count: 84, data: 5600 2472 1626 314  266 716 670 314 670 318 671 315 271 711 273 709 676 315 271 711 668 323  267 713 275 710 673 313 274 709 676 312 276 708 675 310 674 316 273 709  277 706 280 704 280 703 284 704 679 306 679 305 680 310 276 705 282 702  284 700 286[323692] [RFBRIDGE] RF buffer count: 84, data: 5606 2462 1635  304 280 702 682 306 679 307 679 309 278 704 280 704 680 308 279 701 686  301 282 704 284 698 686 303 282 703 682 305 286 699 685 304 681 305 281  702 286 696 290 696 290 693 291 694 690 297 687 302 683 305 281 699 288  698 285 696 291 seems that there is nothing triggering the end of receive and processing.

Thank you for the pulse data !!!
On web UI it is normal to see the log continuing on the same line, using the telnet socket the output is definitely more readable.

I think I can see why rc-switch does not catch your codes ... the library makes some assumptions about the structure of the code, and the leading pulses do not fit in those assumptions. It is something I am seeing with one of my RCs too The good news is that with pulse data it is possible to see how a modified routine performs !

derFrickler, 2018-06-05:
Thanks guys, anything i can help you in fixing this?

derFrickler, 2018-10-17:
Anything i can help you in fixing this?

closed 2018-12-23 by stale[bot]