xoseperez/espurna #916: POW R2 no measurement

nilz90, 2018-06-04:
installed two of them today. Just flashed the pre build images on it. Everything just works fine except power measurement.
Voltage - works fine
Active Power - 0 no changes under load
Energy - 0 no changes under load

any idea?

xoseperez, 2018-06-04:
By default ESPurna will report current and power to 0 if the relay is OFF. Please check if this is your case.

closed 2018-08-13 by xoseperez

gabrielklein, 2018-08-24:
Same issue, with the CSE7759B Pow R2 2017-11-30. Everything is reported as 0.
Please note that switching it on/off has no effect on the module.
Pressing the button on the module switch it on/off but is not reported to the web ui.

I work on the latest version using PlatformIO. I have a PowV1 and it's working perfectly.

jjviton, 2019-11-09:
Hi,

I´m facing a similar issue, I´m not able to read the Power paramater of my flashed POW.

Enviroment: SONOFF POW R2 // Espurna version 1,13 // sensor chip CSE7759

I´ve read in other thread about change parameter using terminal. 

Could you please advise me how to fix this issue?

Thanks in advance.