xoseperez/espurna #917: ARILUX® AL-LC11

mamasta, 2018-06-04:
Hi,
Is it possible to connect 5 pin led strip lights set to ARILUX® AL-LC11
an how ?
The cable I see has 5 male pins and the ARILUX® AL-LC11 has 6 pins.
Please help, do I need an adaptor to buy also???
Thank you
Best regards
Emmanuel

mamasta, 2018-06-04:
[image]

ColinShorts, 2018-06-05:
I think these have two channels for white (warm white and cold white), you will need to connect the cables to the corresponding socket, or alter the code to suit the different colour ordering. Ensure that the common wire is in the correct location - seems to have a 💓 symbol or a red triangle in your picture.

I think the red and green wires need to be swapped, and I'm assuming black is common, but please check.

mamasta, 2018-06-05:
Thank you for your answer.
Yes, the controller is for RGDW & RGBWW.
Black is  - 
Red is    +
The rest C W B R G and the RED triangle, you can see everything in the photo below.

[image]

Sorry, but I can't understand how am I suppose to connect the cables to the corresponding socket?
You mean to cut along each cable with his pin and place each of them to the correspondind socket
to the controller???

Also, common you mean the minus (-) ?

There is not any adaptor to make the 5 pins on the LED strip 6 or a 6pin male to 5pin female adapter?

Sorry for the questions but I'm really lost and thank you again for your time.

Best regards
Emmanuel

ColinShorts, 2018-06-05:
The easiest long term solution would be to swap the red and green wires. Carefully cut off the heat shrink and use a soldering iron to melt the solder to allow you to pull out the wire. Swap the wires round and put on a new bit of heat shrink.

If you can't solder, you can change the color pin assignment in the code.

I'm not aware of any adapters, but I haven't looked for them either...

Black on the led wires is probably common (shared) between RGB+W. Red would be the red LED, Blue is blue, green is green. You can check the polarity of the strip with a multimeter, or by looking at the datasheet for the strip.

mamasta, 2018-06-05:
Thank you so much for your instructions.

But even then, it is not possible to fit the wire's male pins 
in the controller's female holes.

The coloring as you can see, I think it is correct.

Black wire in the red triangle, green in G, red in R, blue in B and white in W.

All are correct. Of Course, I don't know what is the C stands for.

You can see in the attached photo, only the black wire fits correct,
the rest pins, are far apart from the holes.

I'll wait for your instructions, thank you so much.
[image]

ColinShorts, 2018-06-05:
Now I see the problem with the pins. The pitch is different. If the pins on the LED's look as though they will fit snuggly, it might be best to remove the white heat shrink, carefully separate the pins and trim a little material from the black insulation and insert the pins in the correct location. I'd suggest using an epoxy to hold the pins together.

If the pins don't look like they'll be a good fit, you'll need to create an adapter of your own, or solder the wires to the first 4 pins plus one of the 5th or 6th on the header that was supplied with the lc11 module.

Unless anyone has a better idea?

C is cold white, W is warm white.

mamasta, 2018-06-05:
Sorry for the trouble my friend, I really appreciated.

You have been very helpfull.

I wish, this was only a matter of an adaptor and that's it,
but now I have to do a lot of work.

I hope it work, I don't know how they sale
products which they don't match.

Thank you again, you are very kind and patient.

Emmanuel

ColinShorts, 2018-06-05:
I did a quick spot of googling and found these:

http://www.proto-advantage.com/store/product_info.php?products_id=3800079 (5 pin)
http://www.proto-advantage.com/store/product_info.php?products_id=3800080 (6 pin)

If you have access to a soldering iron and make neat work of it, these should be perfect for what you need, but I dare say the shipping might be prohibitive. Aliexpress offers an 8 pin version, but doesn't seem to carry 5/6 pin versions:

https://www.aliexpress.com/item/5-PCS-1-27MM-2-0MM-2-54MM-8pin-Adapter-Board-Adapter-for-Wireless-Module/32322362494.html

That said, in most cases I'd imagine that you'd prefer to have a bit of a gap between the module and the strip, but you would likely be able to pickup 2.54mm cabling a bit easier than an adapter of the same length.

HTH.

p.s. yes, it does seem strange that you'd be sold these two items together, I went down the H801 route which had screw terminals - much simpler!

mamasta, 2018-06-05:
Exactly in this point, in this message of yours 
and with all those measurements,
you have just lost me.

The only thing for sure, is that the pitch between the pins
in the cable......... is 2,5 mm and
in the controller... is 2.0 mm.

I can see that there is some of those items with this
measurments, in the sites you have sent me.

Anyway, I just start with a complaine to the people
who provide those products, because the sale have been made,
after their info and guidance.

Let's see where we will end up.

For one more time I like to thank you.
I can realy see that you love what you are doing,
you are very lucky!

Best regards
Emmanuel

closed 2018-08-20 by stale[bot]