xoseperez/espurna #923: build fails with WEB_SUPPORT=0

copyrights, 2018-06-10:
I was trying to build a firmware with SSL support (`ASYNC_TCP_SSL_ENABLED=1`) and therefore I disabled web support (`WEB_SUPPORT=0` see issue #64) .

The build fails (`master` and `dev`) because other modules refer to web server functions. I had to disable NOFUSS and HOMEASSISTANT as well (`NOFUSS_SUPPORT=0` `HOMEASSISTANT_SUPPORT=0`)

xoseperez, 2018-06-11:
Fixed in dev. Staged for release.

closed 2018-06-11 by xoseperez

ankycooper, 2019-01-06:
I tried both master & dev, but the build fails with web_support=0 works fine with web support.
I'm trying to get secure mqtt to work which requires web to be disabled.
I'm using PLATFORMIO

Here is the log

Compiling .pioenvs\nodemcu-lolin\lib297\ESPAsyncTCP\AsyncPrinter.cpp.o
C:/Users/XXXXXX/espurna/code/espurna/alexa.ino: In function 'bool _alexaBodyCallback(void*, uint8_t*, size_t, size_t, size_t)':
C:/Users/XXXXXX/espurna/code/espurna/alexa.ino:79:33: error: 'void*' is not a pointer-to-object type
alexa.createServer(false);
^
C:/Users/XXXXXX/espurna/code/espurna/alexa.ino:79:52: error: 'void*' is not a pointer-to-object type
alexa.createServer(false);
^
C:/Users/XXXXXX/espurna/code/espurna/alexa.ino:79:66: error: 'HTTP_GET' was not declared in this scope
alexa.createServer(false);
^
C:/Users/XXXXXX/espurna/code/espurna/alexa.ino:79:83: error: 'void*' is not a pointer-to-object type
alexa.createServer(false);
^
C:/Users/XXXXXX/espurna/code/espurna/alexa.ino: In function 'bool _alexaRequestCallback(void*)':
C:/Users/XXXXXX/espurna/code/espurna/alexa.ino:87:27: error: 'void*' is not a pointer-to-object type
#if RELAY_PROVIDER == RELAY_PROVIDER_LIGHT
^
C:/Users/XXXXXX/espurna/code/espurna/alexa.ino:87:62: error: 'void*' is not a pointer-to-object type
#if RELAY_PROVIDER == RELAY_PROVIDER_LIGHT
^
C:/Users/XXXXXX/espurna/code/espurna/alexa.ino:89:33: error: 'void*' is not a pointer-to-object type
// Global switch
^
C:/Users/XXXXXX/espurna/code/espurna/alexa.ino:89:52: error: 'void*' is not a pointer-to-object type
// Global switch
^
C:/Users/XXXXXX/espurna/code/espurna/alexa.ino:89:66: error: 'HTTP_GET' was not declared in this scope
// Global switch
^
C:/Users/XXXXXX/espurna/code/espurna/alexa.ino:89:83: error: 'void*' is not a pointer-to-object type
// Global switch
^
Compiling .pioenvs\nodemcu-lolin\lib297\ESPAsyncTCP\ESPAsyncTCP.cpp.o