xoseperez/espurna #928: Sonoff RF Bridge - moded RF firmware problem

cyanoone, 2018-06-12:
I have problem with Espurna on Sonoff Bridge. Before I flashed modded rf firmware everything was fine. Now I have problem with rf door sensor, it was working good, but now I recive different code every (?) time I open the door. I flashed rf firmware using Tasmota 6.0a (in dev build they added self flashing rf firmware). Could you help me? I am using newest Espurna for rf Bridge.

xoseperez, 2018-06-12:
Maybe @Portisch or @wildwiz might be more helpful here...

xoseperez, 2018-06-13:
I have opened a new issue for this (#935). Can you confirm it is working as expected and close this issue?

cyanoone, 2018-06-14:
@xoseperez After test it seams to work, but my RF remote which partially worked on orginal firmware stopped working. When I click debug I don't get info about any codes. Learning is not working. It is remote from old cheap outlets. I have one more question, could you build Espurna for rf Bridge with rf raw support?

xoseperez, 2018-06-14:
There is already a rfbridge-direct image (https://github.com/xoseperez/espurna/releases/download/1.12.6/espurna-1.12.6-itead-sonoff-rfbridge-direct.bin). Is this what you meant?

wildwiz, 2018-06-14:
Rfbridge-direct does not use the RF mcu at all, so it doesn't care what firmware is on it.
beware, this version of the firmware requres hardware modifications to the board to work.
the details are here https://github.com/xoseperez/espurna/wiki/Hardware-Itead-Sonoff-RF-Bridge---Direct-Hack

cyanoone, 2018-06-15:
Ok, I thought that. I have done firmware mod to make my device more compatible with other devices. Should i do sth to enable better compatibility?

closed 2018-08-24 by stale[bot]