xoseperez/espurna #933: RF Not saving Changes

seetendra, 2018-06-13:
Hi Everyone

I just flashed my RF bridge with Espurna latest version and it works great but for some reason once i have it all setup with my Wifi and MQTT details and power it down and back up again
The Unit just resets it does not save the change 
I have repeated the process 3 times with the same result.
Is there something that iam doing wrong ?

xoseperez, 2018-06-13:
Can you connect to the device serial port (or via telnet), type `info` and copy paste here the output?

seetendra, 2018-06-13:
[059942] [MQTT] Sending RFBridge/button/0 => 1 (PID 1)
[059948] [NTP] UTC Time  : 2018-06-13 19:18:54
[059950] [NTP] Local Time: 2018-06-14 00:48:54
[060006] [MAIN] System OK
[066615] [WEBSOCKET] Requested action: dbgcmd
[066622] 

[066623] [INIT] ESPURNA 1.12.6
[066624] [INIT] xose.perez@gmail.com
[066625] [INIT] http://tinkerman.cat

[066626] [INIT] CPU chip ID: 0xC928D7
[066627] [INIT] CPU frequency: 80 MHz
[066628] [INIT] SDK version: 1.5.3(aec24ac9)
[066630] [INIT] Core version: 2.3.0
[066631] [INIT] Core revision: 
[066632] 
[066633] [INIT] Flash chip ID: 0x144051
[066635] [INIT] Flash speed: 40000000 Hz
[066636] [INIT] Flash mode: DOUT
[066637] 
[066638] [INIT] Flash sector size:     4096 bytes
[066640] [INIT] Flash size (CHIP):  1048576 bytes
[066642] [INIT] Flash size (SDK):   1048576 bytes /  256 sectors
[066643] [INIT] Firmware size:       475344 bytes /  117 sectors
[066644] [INIT] OTA size:            548864 bytes /  134 sectors
[066645] [INIT] EEPROM size:           4096 bytes /    1 sectors
[066646] [INIT] Empty space:          16384 bytes /   4 sectors
[066648] 
[066649] [INIT] BOARD: ITEAD_SONOFF_RFBRIDGE
[066650] [INIT] SUPPORT: ALEXA BROKER DEBUG_TELNET DEBUG_WEB DOMOTICZ HOMEASSISTANT MDNS_SERVER MQTT NTP SCHEDULER TELNET TERMINAL THINGSPEAK WEB

[066667] [INIT] Last reset reason: Reboot from web interface
[066668] [INIT] Settings size: 1669 bytes
[066670] [INIT] Free heap: 10184 bytes
[066674]

seetendra, 2018-06-13:
Please do let me know if your looking for any other info.
I have the device connected to and configured

xoseperez, 2018-06-13:
What if you log this process: type `keys` in the terminal, change something in the web interface and click `save`, then type `keys` again in the terminal. Are your changes there? And after rebooting?

seetendra, 2018-06-13:
Before Making Changes
[016360] [WEBSOCKET] Requested action: dbgcmd
[016378] Current settings:
[016380] > adminPass => Alpha@123
[016381] > alexaEnabled => 1
[016383] > apiEnabled => 0
[016384] > apiKey => 2F2522ED378A7C83
[016385] > apiRealTime => 0
[016386] > board => 26
[016388] > boardName => ITEAD_SONOFF_RFBRIDGE
[016389] > btnDelay => 500
[016390] > btnGPIO0 => 0
[016392] > cfg => 3
[016393] > dczEnabled => 0
[016394] > dczRelayIdx0 => 0
[016396] > dczRelayIdx1 => 0
[016397] > dczRelayIdx2 => 0
[016399] > dczRelayIdx3 => 0
[016400] > dczRelayIdx4 => 0
[016401] > dczRelayIdx5 => 0
[016498] +OK

After Making Changes

[084486] Current settings:
[084488] > adminPass => Alpha@123
[084490] > alexaEnabled => 1
[084491] > apiEnabled => 0
[084492] > apiKey => 2F2522ED378A7C83
[084494] > apiRealTime => 0
[084495] > board => 26
[084497] > boardName => ITEAD_SONOFF_RFBRIDGE
[084498] > btnDelay => 500
[084499] > btnGPIO0 => 0
[084500] > cfg => 3
[084502] > dczEnabled => 0
[084503] > dczRelayIdx0 => 0
[084505] > dczRelayIdx1 => 0
[084506] > dczRelayIdx2 => 0
[084507] > dczRelayIdx3 => 0
[084509] > dczRelayIdx4 => 0
[084619] +OK

Change Made : Enabled Auto discovery in Hass

seetendra, 2018-06-13:
[214290] [WEBSOCKET] Requested action: dbgcmd
[214310] Current settings:
[214312] > adminPass => Alpha@123
[214313] > alexaEnabled => 0
[214314] > apiEnabled => 0
[214316] > apiKey => 2F2522ED378A7C83
[214317] > apiRealTime => 0
[214318] > board => 26
[214320] > boardName => ITEAD_SONOFF_RFBRIDGE
[214321] > btnDelay => 500
[214323] > btnGPIO0 => 0
[214324] > cfg => 3
[214325] > dczEnabled => 0
[214327] > dczRelayIdx0 => 0
[214328] > dczRelayIdx1 => 0
[214329] > dczRelayIdx2 => 0
[214442] +OK

seetendra, 2018-06-13:
Disabled Alexa here i see the change

xoseperez, 2018-06-13:
Are you using the debug log via the web interface? It looks like there are a ton of missing keys and they should be there. 1.12.6 had a problem with dumping too much data to the web debug log. Can you repeat these tests using telnet?

seetendra, 2018-06-13:
Yes i was doing that with the Debug.
Let me try with Telnet
What is the username and password i should use for telnet?

seetendra, 2018-06-13:
Tried it with out any password no go

xoseperez, 2018-06-14:
You should first enable Telnet in the web interface, in the Admin tab. Once enabled there is no password.

seetendra, 2018-06-14:
Got it working 
Just went ahead and re flashed  it and this time it started to work without any issues 
It is saving the changes made

Iam all set.

xoseperez, 2018-06-14:
All's well that ends well. Closing :smile:

closed 2018-06-14 by xoseperez