xoseperez/espurna #934: 2 BME280 or BME+BMP280 on the same I2C

hyteoo, 2018-06-13:
I tried setting up a espurna with a BMP280 (0x76) and a BME280 with address changed to 0x77 on the same I2C

Unfortunately only the first is detected (0x76). If I only hook up one of the sensors, they both work and addresses are detected correctly.

Is there any change I could make to support both sensors at the same time?

Thanks!

hyteoo, 2018-06-13:
Great, that worked.
What are the sensor types that support more than one?

xoseperez, 2018-06-13:
The DS18B20 uses a one-wire bus, you can connect several of those sensors to the same GPIO and the code will recognize them all.

closed 2018-06-13 by xoseperez

einnordlicht, 2019-03-09:
I have succeeded to get 2 BMP280 running with the above phows humitatch, one on 0x76 and one 0x77 (by pulling SDO on Vcc). However even if the sensors support temperature, humidity and pressure, only the first BMP280 shows humidity. If I exchange both sensors, again the first shows humidity, and the other not. So in general both sensors are proven to work, but only one humidity value is shown.