xoseperez/espurna #936: How to activate RF on 4ch pro r2

fpalinf, 2018-06-13:
Is there a way to activate RF on sonoff 4ch pro r2 ?

I flashed with espurna-1.12.6-itead-sonoff-4ch-pro.bin and cannot find the option.

Or should I rebuild with some specific flags ?

Regards,

xoseperez, 2018-06-13:
Not 100% sure, I have not tested the R2, but I'd say the RF is handled by the secondary chip, so there is nothing to do from the ESPurna point of view. Pairing and toggling via RF should work the same way it does with the stock firmware.

Check the wiki page for the device: https://github.com/xoseperez/espurna/wiki/Hardware-Itead-Sonoff-4CH-Pro---R2

fpalinf, 2018-06-13:
My mistake. I thought the rf433 on ch4 was equivalent to the rf433 bridge.
From what I understand the board can only receive and not send commands. 
Would it be still possible to have the same learning capabilities than we see on thr rf433 bridge ? (To see the learned frame ?)
If not how to dissociate the button so getting the rf433 frame will not trigger the relay command ?

xoseperez, 2018-06-14:
No, the secondary chip only reports the equivalent to button presses and relay statuses, no frames. The only way is to modify the firmware in that chip to make it work like a transparent bridge and handle all logic in the ESP8266. Very much like what has been done with custom firmware for the EFM8 in the RF Bridge. But that's a whole project...

closed 2018-08-20 by stale[bot]