xoseperez/espurna #937: web ui does not work over HTTPS when connecting from outside LAN with reverse proxy

hyteoo, 2018-06-14:
I currently run all my espurna devices inside a separate LAN, with no forwarding to WAN.
My setup involves a pfsense router with reverse proxy for accessing everything I need inside the LAN, over HTTPS only.

From time to time however I need to do some admin over the internet, so I have setup a reverse proxy rule that I can easily change to point to a specific espurna device.

Problem is when connecting over HTTPS, rev.proxy connects over HTTP to the espurna. Page is served but when webservices are called, they are called over HTTP (ws://....) so they end up nowhere.

Error received after logging in:

`Mixed Content: The page at 'https://xxxxxx.com/' was loaded over HTTPS, but attempted to connect to the insecure WebSocket endpoint 'ws://xxxxxx.com/ws'. This request has been blocked; this endpoint must be available over WSS.`

A workaround could be to detect if we are HTTP or HTTPS and request the WS accordingly. 
What do you think?

xoseperez, 2018-06-14:
Yes, it's enough with changing the protocol accordingly. I have updated the code in dev branch. Here's my working configuration using NGINX:

```
    location /espurna/ {

      # header pass through configuration
      proxy_set_header        Host $host;
      proxy_set_header        X-Real-IP $remote_addr;
      proxy_set_header        X-Forwarded-For $proxy_add_x_forwarded_for;
      proxy_set_header        X-Forwarded-Proto $scheme;
      proxy_set_header        Upgrade $http_upgrade;
      proxy_set_header        Connection "Upgrade";

      # extra debug headers
      add_header              X-Forwarded-For $proxy_add_x_forwarded_for;

      # actual proxying configuration
      proxy_ssl_session_reuse on;
      proxy_pass              http://espurna-b93c93.local/;
      proxy_redirect          default;
      proxy_read_timeout      90;

    }
```

closed 2018-06-21 by xoseperez

lblabr, 2019-07-14:
do you also have an example for an reverse proxy configuration for apache ? i don't get the ws running...

maybe i have a little complex setup
-> apache reverse proxy -> haproxy -> espurna device

http access to backend behind happroxy are working, i also get the espurna login but nothing else an empty page...

could you give me some advice ?

thanks in advance

Lars