xoseperez/espurna #938: Any plans for integration with RFM69 (MQTT RFM69 gateway)?

hyteoo, 2018-06-14:
I love the RFM69 gateway you built (http://tinkerman.cat/rfm69-wifi-gateway/)
I am building sensors based on RFM69, currently using a ESP8266-RFM69 gateway with some custom software I wrote, however I'd love to also have espurna on that.
I saw the modded espurna you provide in the article, however that's an old version and it's come a long way since then.

Any chance to integrate support into the official espurna for that?
I know there's the sonoff gateway, however the rfm69 features dwarf the simple RF modules used in the sonoffs, and also it's more tinker-able :)

I'd be happy to help as I'm actively developing on the topic.

xoseperez, 2018-06-14:
There are no plans at the moment. Problem is that the RFM69GW has some additional web interface features and I don't have a way to (easily) build custom web interfaces in ESPurna, like removing parts of the web interface that are not going to be used for a certain device.

But I do have plans to update the RFM69GW project at some point, maybe even porting it to ESP32.

hyteoo, 2018-06-14:
That would be great
Even though a big part of espurna is supporting existing commercial devices, I think it's potential is in using it on custom devices. And a RF based setup is a must in any smart home... relying on wifi alone is not very efficient.

Hope you get to move the project here on github and update it soon :)

knopserl, 2018-06-17:
@xoseperez. I'm working on varios ESP32 solution (ASkSin, LoRa), so I'm interested also in other projects with ESP32. I think the ESP32 (the small bare ESP32S board for 3€) is really a nice MCU. However I like the RFM95 more than the RFM69.

knopserl, 2018-06-18:
@xoseperez thats perfect, I like that combination: ESP32 + RFM95. RFM69 is a nice to have but RFM95 is definitly the best option. I'm not planning to participate in the TTN but rather build a LoRaLAN instead of a LoraWAN, so a gateway to my local MQTT server and then NodeRed of even OpenHAB, but I think I can reuse the project for that purpose, right? Or install a local TTN server?

xoseperez, 2018-06-24:
OK, I have made up my mind... check the `rfm69` branch :see_no_evil:

svwhisper, 2018-07-12:
It's great to see this functionality added.  Can I suggest that when Espurna detects it is running as a gateway that RFM69 ATC is disabled?  I think different RFM69 nodes present with quite different RSSI and therefore ATC on the gateway instance of Espurna is likely a hindrance rather than a help.  I would like to be able to propose the code changes required, but am not yet skilled enough to do so.

xoseperez, 2018-07-12:
Actually, it was not enabled but the debug message was wrong. Anyway, I have disabled it explicitly now.

closed 2018-07-26 by xoseperez