xoseperez/espurna #939: Sonoff POW R2 - Mqtt error

arek156, 2018-06-14:
Hello
I have a problem with sonoff pow r2, after uploading espurna everything works until the mqtt client starts. After turning on the mqtt, i can see a significant delay in switching, additionally at various intervals, the sonoff is switched off and finally the wifi network is lost. I use other Sonoff S20 devices and espurna software and everything works, but not pow r2, I have the latest of espurna soft 1.12.6
Do you have any idea what can cause mqtt error with this switch?

Regards
@rek

xoseperez, 2018-06-14:
Is it a local MQTT broker? What configuration are you using? It could be helpful to see the debug log while sending MQTT commands to the device.

arek156, 2018-06-14:
local mqtt broker, installed on raspberry pi, port 1883

xoseperez, 2018-06-14:
Can you share a debug log showing the issue?

arek156, 2018-06-14:
[log_sonoff_pow2.txt](https://github.com/xoseperez/espurna/files/2102962/log_sonoff_pow2.txt)

arek156, 2018-06-14:
Is this log correct?

arek156, 2018-06-14:
Hi, I've uploaded the software again today. This time mqtt works, however, after switching the sonoff to ON, it automatically switches to OFF at different time intervals.

xoseperez, 2018-06-15:
Questions/issues:
* I can only see a relay change request from the web interface
* I don't see any relay OFF request in the log
* Could it be your domoticz turning the relay off?
* Looks like your MQTT root topic is not set.
* Sending a temperature value every 2 seconds is a little overkill, isn't it?

xoseperez, 2018-06-16:
You can set the root topic in the MQTT tab of the web interface. You don't have pulses enabled, do you?

arek156, 2018-06-16:
Hi, I completed the data in mqtt root topic, however it does not help. I noticed that the problem only occurs if I enable mqtt support. I am using sonoff bridge and sonoff s20 on your software and everything works, so I put a problem on the sonoff pow r2 side. I also installed espurna on sonoof basic and it also works perfectly, I will try to clear the memory and upload the soft one again, maybe it will help.

P.S I don't have pulse enabled

arek156, 2018-06-24:
Hi
After update to 1.13.0 everythings works great

xoseperez, 2018-06-24:
Cool! I'm closing this issue then. 
Thank you for the feedback.

closed 2018-06-24 by xoseperez