xoseperez/espurna #941: ESPurna duda

Camposmansi, 2018-06-14:
Buenas, tengo un ESP Homcube y le estoy metiendo el generic, pero no consigo que me funcione el boton, ni el medidor de corriente
¿Cual hay que pueda meterle que sea mas generico?

xoseperez, 2018-06-15:
Tienes  las especificaciones de ese dispositivo?

xoseperez, 2018-06-16:
> Para que lo veas es este,
>
> https://www.androidtoplay.es/tienda/varios/enchufe-inteligente-wifi-salida-del-zocalo-del-interruptor-del-temporizador-funciona-con-amazon-alexa-y-google-homeno-se-requiere-hubcontrole-sus-dispositivos-desde-cualquier-lugarpara-android-ios/
>
> Buscando especificaciones detenidamente me acabo de dar cuenta, que hay
> muchas marcas que usan el mismo exactamente como uno de BlitzWolf
>
> https://www.amazon.es/Inteligente-Temporizaci%C3%B3n-Dispositivos-Cualquier-BlitzWolf/dp/B07BMQP142/ref=sr_1_9?ie=UTF8&qid=1529065194&sr=8-9&keywords=enchufe+google+home
>
> Los gpios que corresponden:
> GPIO0 Led Rojo. Inverso en espurna: Off Encendido
> GPIO2 Led Azul. Inverso en espurna: Off Encendido
> GPIO13 Boton.
> GPIO15 Rele.
>
> En el contador de consumo, me pierdo, probare con el sonof POW y te
> comento si lee algo

Mmm... Interesante. Acabo de pillar uno para trastear con él.

Camposmansi, 2018-06-16:
Jajajaja, ya me mantienes informado, much as gracias

Camposmansi, 2018-06-21:
wow no lo vi, este finde lo pruebo, muchas gracias por todo, eres genial