xoseperez/espurna #944: wiity cloud ap issue

unique904, 2018-06-14:
im unable to get the witty cloud to work with any espurna fw as well as the specific gizwiz fw
there looks to be an issue with the wifi AP
[000407] [WIFI] MODE AP --------------------------------------
[000407] [WIFI] SSID  ESPURNA_######
[000410] [WIFI] PASS  fibonacci
[000413] [WIFI] IP    0.0.0.0
[000415] [WIFI] MAC   ##:##:##:##:##:##
[000419] [WIFI] ----------------------------------------------
[000425] [MDNS] FAIL
[060003] [MAIN] System OK

xoseperez, 2018-06-15:
Do you mean it does not work with any of the two firmware images? Have you tested a simple wifi sketch?

unique904, 2018-06-15:
Just tested the pre released bin files including the core 1mb and core 4mb
Not got as far as using sketch’s

I have loaded espeasy on to it with out any issues so does seam to b a problem with the .bin release
Kind Regards

Craig Bennett

swilson86, 2018-06-15:
I was unaware of pre released bin files, but it brings up an interesting point. They should be labeled Alpha or Beta bin files and some of us can quickly test them to ensure that fixes and enhancements are working prior to a major release. It would be a disaster should a new release fail everyone.

unique904, 2018-06-15:
Sorry that’s my mistake I meant to say the prebuilt release https://github.com/xoseperez/espurna/releases/

Kind Regards

Craig Bennett

xoseperez, 2018-06-16:
Yes, there are no pre-release binaries (so far) and if there were I would probably not publish them in the same place as the "stable" releases.

swilson86, 2018-06-16:
Hmmmm Xose. Your caveat "so far" is interesting. I am anxious for example to test the fix on the S31 power measurements. I check daily like a child on Christmas morning to see if 1.12.17 is ready. If I knew how to compile pre-releases or test Beta/Alpha binaries that would be wonderful. This typo mentioning pre-releases brings up an excellent point.

mcspr, 2018-06-17:
@xoseperez @swilson86 Funny you should ask... builds for issue at hand (dev): https://github.com/mcspr/espurna-travis-test/releases/tag/20180617.3

It was a cloned repository to test parallel jobs for travis releases, but they are still janky for building large amount of images (93 already!) because they cannot easily share already built webui / arduino installation / arduino libraries. And there is an earlier try at https://github.com/mcspr/espurna-travis-test/releases/tag/20180617, until the cache situation gone wrong and it stopped building mid-process.

closed 2018-08-23 by stale[bot]