xoseperez/espurna #945: compiling 1.12.6

jonny190, 2018-06-14:
I get the following issues:

C:\Users\Jonny\Documents\espurna-master\code\espurna\wifi.ino: In function 'void _wifiConfigure()':

wifi:28: error: 'class JustWifi' has no member named 'setAPMode'

     jw.setAPMode(WIFI_AP_MODE);

        ^

In file included from sketch\config/all.h:31:0,

                 from C:\Users\Jonny\Documents\espurna-master\code\espurna\espurna.ino:22:

sketch\config/general.h:280:37: error: 'AP_MODE_ALONE' was not declared in this scope

 #define WIFI_AP_MODE                AP_MODE_ALONE

                                     ^

C:\Users\Jonny\Documents\espurna-master\code\espurna\wifi.ino:28:18: note: in expansion of macro 'WIFI_AP_MODE'

     jw.setAPMode(WIFI_AP_MODE);

                  ^

wifi:59: error: 'class JustWifi' has no member named 'scanNetworks'

     jw.scanNetworks(getSetting("wifiScan", WIFI_SCAN_NETWORKS).toInt() == 1);

        ^

C:\Users\Jonny\Documents\espurna-master\code\espurna\wifi.ino: At global scope:

wifi:201: error: 'DNSServer' does not name a type

 DNSServer _wifi_dnsServer;

 ^

C:\Users\Jonny\Documents\espurna-master\code\espurna\wifi.ino: In function 'void _wifiCaptivePortal(justwifi_messages_t, char*)':

wifi:206: error: '_wifi_dnsServer' was not declared in this scope

         _wifi_dnsServer.setErrorReplyCode(DNSReplyCode::NoError);

         ^

wifi:206: error: 'DNSReplyCode' has not been declared

         _wifi_dnsServer.setErrorReplyCode(DNSReplyCode::NoError);

                                           ^

wifi:212: error: '_wifi_dnsServer' was not declared in this scope

         _wifi_dnsServer.stop();

         ^

C:\Users\Jonny\Documents\espurna-master\code\espurna\wifi.ino: In function 'bool createAP()':

wifi:376: error: 'class JustWifi' has no member named 'createAP'

     return jw.createAP();

               ^

C:\Users\Jonny\Documents\espurna-master\code\espurna\wifi.ino: In function 'void wifiLoop()':

wifi:475: error: '_wifi_dnsServer' was not declared in this scope

             _wifi_dnsServer.processNextRequest();

             ^

exit status 1
'class JustWifi' has no member named 'setAPMode'

xoseperez, 2018-06-15:
ESPurna 1.12.6 uses JustWiFi 1.2.0, you are probably fetching JustWifi 2.0.0 which has some changes and it's not backwards compatible.

jonny190, 2018-06-16:
ive dropped to this but geting the following 

wifi:201: error: 'DNSServer' does not name a type

 DNSServer _wifi_dnsServer;

 ^

C:\Users\Jonny\Documents\espurna-master\code\espurna\wifi.ino: In function 'void _wifiCaptivePortal(justwifi_messages_t, char*)':

wifi:206: error: '_wifi_dnsServer' was not declared in this scope

         _wifi_dnsServer.setErrorReplyCode(DNSReplyCode::NoError);

         ^

wifi:206: error: 'DNSReplyCode' has not been declared

         _wifi_dnsServer.setErrorReplyCode(DNSReplyCode::NoError);

                                           ^

wifi:212: error: '_wifi_dnsServer' was not declared in this scope

         _wifi_dnsServer.stop();

         ^

C:\Users\Jonny\Documents\espurna-master\code\espurna\wifi.ino: In function 'void wifiLoop()':

wifi:475: error: '_wifi_dnsServer' was not declared in this scope

             _wifi_dnsServer.processNextRequest();

             ^

exit status 1
'DNSServer' does not name a type