xoseperez/espurna #947: Sonoff Pow v2.0

analistamarcioprado, 2018-06-15:
Good morning people!

I bought 4 Sonoff Pow, but came to version 2.0.

It does not have the pins exposed to upload the firmware.

In addition to other small visual differences of the board.

Has it ever managed to put the ESPurna in a Sonoff Pow v2.0?

Here is the equipment photo:

[image]

[image]

Thank you very much in advance.

erkgh, 2018-06-15:
Hi,

I am almost sure that the 4 holes in  the left bottom corner of your first picture are the pins needed for flashing firmwares.

GND, RX, TX, VCC (3.3V).

After making the proper connections the procedure  should be the same as for the other/older models.

Good luck, cheers

Eric

analistamarcioprado, 2018-06-15:
Hi,

That's right.

My concern is that the hardware is different ...

xoseperez, 2018-06-16:
The POW 2 is supported, solder some headers to those 4 pins and connect a
programmer, press and hold the button an power the board to boot into flash
mode.

seetendra, 2018-06-17:
I have the Same Sonoff POW and they are supported .
Just flash them the same way you flash any sonoff

Here is a Video by Dr z 

https://www.youtube.com/watch?v=BCEU8Ijmkds&t=6s

analistamarcioprado, 2018-06-17:
Thanks for the contributions.

I did not solder the pins. I fit in a way that made contact. But I must have done something wrong.

My Sonoff when connected via USB does not turn on any led.

When it is connected to the mains, there is a period of time with the LED on for a long time and then off.

Is there any possibility of recovering the equipment?

seetendra, 2018-06-18:
I dont think your Sonoff is bad 
Believe me it is very difficult to do that. I have done all kinds of stuff with the POW V2 that i have and they are all working fine

Just make sure that your flashing them with 3.3v on the FTDI adapter and not 5v 

Also use the ESPEASY method that is the fasted and you don't have to Hold on to the Sonoff With the Pins for a very long time

I never solder pin to the Sonoff that i flash there is no need to do that just hold the pins in place for some time and there should be no problem.

Wiring the FTDI adapter is al important make sure you have that correct. 
[image]

seetendra, 2018-06-18:
[image]

analistamarcioprado, 2018-06-18:
Thanks for the info.

The pinning issue I believe I did not make a mistake. I used this one (although it is version 1.0):
[image]

How is this method called ESPEASY?

schweini, 2018-06-21:
Be very careful with the Sonoff POW, because the internal pins can sometimes be at 110V/220V relative to earth! So not connect it to mains volltage while flashing it!

analistamarcioprado, 2018-06-22:
Staff thank you all.

The problem was very simple. More for a beginner in Sonoff Pow gave a lot of work to find out.

The problem is the sequence of the pinout.

All the tutorials I saw in the sequence were this:

GND
TX
RX
TRUE

But is the correct sequence in Sonoff Pow ver 2.0 the following?

GND
RX
TX
TRUE

Hugs.

[image]

analistamarcioprado, 2018-06-22:
Guys, I think I said something wrong.

Does the TX di Sonoff pin have to be connected to the RX or TX of the USB interface?

ColinShorts, 2018-06-22:
If it doesn't work one way, switch RX and TX leads and try again. As long as you connect the negative and positive leads the correct way and use the correct voltage (usually 3.3V) it'll be fine.

analistamarcioprado, 2018-06-22:
Ah yes. Nice!

I did this and it really worked.

Once again thank you for your attention.

analistamarcioprado, 2018-06-24:
Please, can put as solved.

xoseperez, 2018-06-25:
Cool

closed 2018-06-25 by xoseperez