xoseperez/espurna #948: COMMERCIAL USE

tcpipchip, 2018-06-15:
Sir
I would like to use the SONOFF in a HOME AUTOMATION product with your software, not the original software by sonoff
Will i have legal issues if i use it ?
Or, if i use with sonoff modules, will i have legal issues with Sonoff manufacturer ?

Thank you!

xoseperez, 2018-06-16:
ESPurna is GPL-3.0 which means you can use it for commercial products and sell it as long as you provide the source code for your product in the same terms as here. So if you modify it to match your specific product requirements (look and feel, new or modified features, product configuration,...) your modifications must be available for anyone under GPL-3.0. 

As for the legal issues with IteadStudio. Not really sure they would like it, but you should contact them anyway.

iDeam0316, 2018-06-17:
Do I have to make the source code public even if I only modify the general.h to change the default setting ?

TMaYaD, 2018-06-17:
Yes. IANAL.

Geroldswil, 2018-06-18:
### Liability
This might be another point you have to be very careful. If you sell modified Sonof products, you have to provide warranty etc. according to your location. The certificates (CE etc.) will loose their validity which means you could be made responsible for any damage caused by these products...

closed 2018-08-24 by stale[bot]

OttoWinter, 2019-02-23:
Just as a tip (ran into this myself): Technically distributing an builds of this software violates GPLv3

Why? GPLv3 is a copyleft license, so all sources this project uses must be available publicly too. This includes the ESP SDK used by the arduino core. But the ESP SDK is closed-source, so linking it in statically is against GPLv3. (dynamic linking would correct, but that's not possible with these devices)

Morefo here: https://github.com/espressif/ESP8266_RTOS_SDK/issues/1 (issue is about rtos sdk, but applies to nonos sdk aswell)

mcspr, 2019-02-24:
*maybe lie, dont quote me* IANAL, but I think there is a section in Gpl3 where it states about possibility to include closed library when author gives permission (which, "i think", current nonos license does). Old SDK did include GPL'ed headers, but current one no longer does it and is purely BSD.

Additional discussions raised another issue - ESP8266_new_pwm code, which is licensed under GPL-2:
https://github.com/SmingHub/Sming/issues/1397
https://github.com/StefanBruens/ESP8266_new_pwm/issues/16
ESPurna includes it: https://github.com/xoseperez/espurna/blob/dev/code/espurna/pwm.c

OttoWinter, 2019-02-24:
(for clarification: I am also not a lawyer, and this is also very new ground for me, so my comments may not be accurate)

So I read more about GPLv3, and from what I have seen you appear to be correct (if the nonos sdk can be considered a "system library" which it probably is)

In fact the gplv3 faq even lists that question:  https://www.gnu.org/licenses/gpl-faq.html#SystemLibraryException (the license of the work doesn't appear to matter, at least from what I understand in that clause)

I have read through some parts of that faq and have only found this for the exception: https://www.gnu.org/licenses/gpl-faq.html#GPLIncompatibleLibs (which sould require manual addition to the license)

Anyway, from what I see the system library exception seems to apply - so this is not an issue after all :)