xoseperez/espurna #950: Travis pre-release binaries

mcspr, 2018-06-18:
As mentioned in #944 and #896 - there is a way to build `dev` branch and release them using CI services.
Currently testing via Travis and Gitlab (... also contemplating CircleCI)

https://github.com/mcspr/espurna-travis-test
https://travis-ci.org/mcspr/espurna-travis-test/builds/393185548

https://gitlab.com/mcspr/espurna-travis-test
https://gitlab.com/mcspr/espurna-travis-test/pipelines/24078032

Parallel builds were the a test at first, because 50 minute builds were failing. Ofcourse, they introduce latency by separating each job into different containers (although they start up pretty fast)

Current implementation for travis triggers builds on tag (date for verbosity) and simply fetches current dev HEAD. But it still needs a way to keep built webui - jobs need to have common node_modules and index.html.gz.h, so it requires some external cache. Internal travis cache is not reliable enough as can be seen in the middle of travis build log (job 39 and further)

All CI services support triggering build on demand, so it would be possible to pass some variables to allow building specific commit, and maybe change hardcoded platform 1.5.0/1.6.0/1.7.1 etc. by editing env file directly.

mcspr, 2018-06-29:
Kudos to #986, there is saner way to build releases. So scratch OP part about totally parallel gitlab/e-t-t repos. 

$TRAVIS_REPO_SLUG is changeable - [releases deploy script](https://github.com/travis-ci/dpl/blob/master/lib/dpl/provider/releases.rb#L40) uses it and happily deploys wherever auth token allows. Triggering can be scheduled based on time (first build + every 24 hours, 7 days or 1 month) via travis cron or using it's API and doing manually.

Because `releases` provider is bound to git tags, it is either `$(date)-$(git rev-parse --short HEAD)` or single `latest` with some name mangling to overwrite old files.

Removing old releases either from web or using api (maybe through deploy key and additional script for after_deploy stage)

mcspr, 2018-07-10:
Builder repo now has 2 working implementations for gitlab and travis.

Scheduling builds for gitlab is more direct - syntax is simple cron (0 7 * * * is 7:00 every day) but builds on are much slower than travis - [23min](https://travis-ci.org/mcspr/espurna-travis-test/builds/402353371) vs [53min](https://gitlab.com/mcspr/espurna-travis-test/pipelines/25418527). Firmware build time is twice as long. Maybe need to try non-shared runners...
https://gitlab.com/help/user/project/pipelines/schedules

Travis cron feature does require web interaction / API call at the **exact** time build must be ran.  And after some time it might drift.
https://github.com/travis-ci/travis-ci/issues/8705
https://developer.travis-ci.org/resource/cron#create

As for on-demand builds, what comes to mind is to use https://developer.travis-ci.org/resource/requests#create "message" part (which will be $TRAVIS_COMMIT_MESSAGE) either as an argument to the build.sh, or replace internal $environments or build.sh with it's value. I wish github API allowed attaching files to issues. So output is either github releases to some other repo or different file storage altogether.

xoseperez, 2018-09-09:
Not stale. I think it will be interesting to have this made official and promote it from the project home page. It could be your repo, doesn't matter.

mcspr, 2018-09-10:
Yes. It is working ok'ish right now.
Minor problem is https://github.com/travis-ci/dpl/issues/229 - sometimes job fails at deploy step because of 502 error from github side. For now probably will just add [ghr](https://github.com/tcnksm/ghr) to the script, it does implement retries.
And it needs minor rework to tag commits on separate branch instead of the main one.

mcspr, 2018-09-11:
Sorry, I meant builder repo itself. See: https://github.com/mcspr/espurna-travis-test/commit/ba12d4726a502a8f7e10820764c36136244162e5
`...` reveals long list of tags <-> releases, which i want to avoid and instead attach release to separate commit on a different branch, made when travis job is started.

Regarding branches here, maybe it would actually be nice to also build some of softcfg. Maybe just one generic d1 env.

xoseperez, 2018-09-19:
So good. Nothing against the name. I would use the name nightly plus date for the nightly dev builds. I think people are more used to that naming convention.

mcspr, 2018-09-21:
Mm something like this then. Replacing version with extended one.
[`VERSION_FMT = "{version}.nightly{tag}.git{sha}"`](https://github.com/mcspr/espurna-nightly-builder/blob/dd117cd08e175b1cb7851b0126917e33ec702c34/espurna_nightly_builder/rename_releases.py#L5)

*Or* if adhering to semver:
https://semver.org/#spec-item-9
`1.13.3a` becomes ...
`1.13.3-alpha` or `1.13.3-dev.1` (2,3,4 etc... or git-describe commit counter)
`1.13.3-dev.git{sha}+nightly{tag}`

mcspr, 2018-10-16:
Bump? README updated -> https://github.com/mcspr/espurna-nightly-builder#espurna-nightly-builder

Versioning may be a separate issue altogether, but the goal is to distinguish between versions. Maybe it should modify VERSION (version.h) too? Make VERSION always append `-dev` by default and only use plain version when building master branch from travis. And, similarly, append `-nightly` for nightly builder.

xoseperez, 2018-11-19:
According to semver we are free to add any tag after the version for pre-releases. 
I would use specific tags for specific versions, like if we ever freeze development on a release candidate using alpha/beta/rc1/rc2/... tags. For nightly builds I would speficy `nighlty` plus date and gitd, pretty much like you are already doing:

* Release: `espurna-1.14.0-board.bin`
* Specific pre-relase version: `espurna-1.14.0-rc1-board.bin`
* Development images (on demand?): `espurna-1.14.0-branch-20181201-a0b1c2e-board.bin`
* Nightly builds: `espurna-1.14.0-nightly-20181201-a0b1c2e-board.bin

Not sure about the `dev` images, but I guess best practice would be to automatically append the branch name if it's not `master`.

mcspr, 2018-11-24:
Yep. I'd look into how it would make sense deduce / store.

And since I had not touched builder for 2 month and it still working - treating as stable ⚓️
Returning to original topic... README should mention it as Notice? e.g.

> Here-> espurna-nightly-builder/releases link <- is the latest development release of ESPurna, built from the latest changes to the 'dev' branch. It can be used to help development process and test for the upcoming stable release.

mcspr, 2019-03-27:
FYI I disabled them for now, testing some other CI providers.
I'd make this milestone for v2 to keep them smaller in size / quicker to build.

closed 2019-06-02 by stale[bot]