xoseperez/espurna #951: Travis release build time

mcspr, 2018-06-18:
Would it still be a problem for 1.13.0 release?

https://docs.travis-ci.com/user/customizing-the-build#Build-Timeouts
1.12.5 did timeout:
https://travis-ci.org/xoseperez/espurna/builds/363768878
1.12.6 did not (?):
https://travis-ci.org/xoseperez/espurna/builds/373842741

xoseperez, 2018-06-22:
Under test at the moment. Let's see what happens.

xoseperez, 2018-06-23:
It should. I have it installed globally in my system. 
I think its time to start thinking on a different solution... this will only go worse.

mcspr, 2018-06-26:
https://github.com/xoseperez/espurna/pull/986 does approach this with workaround using fixed amount of travis jobs and splitting builds evenly between them.

The main sore is that serial builds waste time on:
- recompiling arduino core each time.
- recompiling each lib_deps
- compiling ALL lib_deps, despite no source dependency. idk if i am missing something about `plus` variants of ldf_mode, but wouldn't they track it? (and they break on Core headers right now, like DNSServer.h and ESP8266mDNS.h)
All of which does add up...

mcspr, 2018-06-26:
Some statistics about pre-built firmware usage.
https://gist.github.com/mcspr/111a7d6addefc75b39de86ba441d92cc

(similar, https://www.somsubhra.com/github-release-stats/)

lobradov, 2018-06-27:
I asked PlatformIO community how to cache some of the frameworks/libraries. 
https://community.platformio.org/t/reusing-compiled-framework-and-libraries/4455

Hopefully Ivan will have an idea.

closed 2018-11-01 by stale[bot]