xoseperez/espurna #952: Crash report is cut!

jsponz, 2018-06-18:
Hi,

after a crash, I run "crash" and the info is not shown completely. 

It is cut:

...
[DEBUG] 3fffedf0: 3fff4aec 3fff3400 3fff113c 40234ed1 
[DEBUG] 3fffee00: 005e0001 0000fb00 ffffff00 ffffffff 
[DEBUG] 3fffee10: 0000001

Ideas?

Thanks!

mcspr, 2018-06-18:
Does this happen with 1.12.6 release .bin or `dev` branch?
Trying to reproduce it with `dev`, no luck so far. Using `factory.reset` + `reset` to fill eeprom with crash data.
It outputs all 66 address lines, both using telnet and serial connection

jsponz, 2018-06-19:
It is my own compilation. 

SDK version: 2.2.1(cfd48f3)

xoseperez, 2018-06-21:
Based on current `dev` branch? There have been some improvements when debugging from telnet or web UI.

jsponz, 2018-06-21:
No, it is based on the master branch.

xoseperez, 2018-06-21:
Try `dev`, as I said there are some improvements in the debug output.