xoseperez/espurna #954: Beta version 1.13.0c hyphen name bug

swilson86, 2018-06-18:
Xose. I am testing 1.13.0c on a couple of S31 switches and see that a space is no longer permitted in the Hostname, but a hyphen is okay. If using Alexa this will fail. 

Example... I named what used to be "Mini Fridge" as "Mini-Fridge" as 1.13.0c forces me to do so. Alexa will now only turn off the device, and not on again. She responds with "Hmmm. I'm not sure about that". I can however turn on the S31  from my iPhone with alexa.amazon.com > Smart Home > Devices if I click on Mini-Fridge, but never by voice.

Suggestion: Allow a space in the Hostname.

xoseperez, 2018-06-19:
With 1.13.0 I'm applying hostname requirements as per RFC (see #874 and #889). Only A-Z0-9 and _ are allowed. 

To allow this from ESPurna there should be an alias for the Alexa device. In the meantime I suggest using the Alexa app to create a group with the device and the required name.

swilson86, 2018-06-19:
After reading the various RFC's you are correct in that a space in a Hostname is not allowed. Fortunately you have a field named MQTT Root Topic which DOES allow a space, thus avoiding a major rewrite of my node-red code. For those upgrading to 1.13.0 they can change {hostname} in that field to whatever the name is using spaces (which is valid in MQTT).

I have setup a Hostname with a Sonoff POW named "Printer-2" and I have noticed Alexa will respond to "Alexa turn on [off] printer 2" but will rarely respond to "Alexa printer 2 on [off]". Therefore it is Amazon that must adapt to the new convention and not Espurna. I will continue my investigation.

closed 2018-06-19 by swilson86