xoseperez/espurna #956: SWITCH#1 on Web Interface controls wrong MQTT TOPIC

brunohorta82, 2018-06-20:
When I press SWITCH #1 de command sent to MQTT is related to relay/0 instead  relay/1

[014582] [MQTT] Sending ESPURNA-A7BDF0/relay/0 => 1 (PID 1)

hardware.h
// Buttons
    #define BUTTON1_PIN         12       // Button 1 on header
    #define BUTTON2_PIN         13      // Physical button
    #define BUTTON1_RELAY       1
    #define BUTTON2_RELAY       2
    #define BUTTON1_MODE        BUTTON_SWITCH | BUTTON_SET_PULLUP | BUTTON_DEFAULT_HIGH
    #define BUTTON2_MODE        BUTTON_SWITCH | BUTTON_SET_PULLUP | BUTTON_DEFAULT_HIGH

    // Relays
    #define RELAY1_PIN          4
    #define RELAY2_PIN          5
    
    #define RELAY1_TYPE         RELAY_TYPE_NORMAL
    #define RELAY2_TYPE         RELAY_TYPE_NORMAL

poulch74, 2018-06-20:
i see same bug in espurna-dev. only #0 relay controls with checkbox on status page, any other added relays generate #0 control in DEBUG log. espurna-master 1.12.6 work fine.

brunohorta82, 2018-06-20:
I see the bug, the function doToggle in custom.js not send de id, the socket only receives a JSON {status:0} instead {id: 1, status:0} or  {id:0, status:0}

brunohorta82, 2018-06-20:
the function in **master** is 
function doToggle(element, value) {
    var id = parseInt(element.attr("data"), 10);
    sendAction("relay", {id: id, status: value ? 1 : 0 });
    return false;
}

**in DEV**

function doToggle(id, value) {
    sendAction("relay", {id: 1, status: value ? 1 : 0 });
    return false;
}

xoseperez, 2018-06-20:
I can see there is a problem with the new checkbox slides when using multiple relays but that code you are posting is not the code in dev...

https://github.com/xoseperez/espurna/blob/b56e986e69125643a8b08299e049ede2459346a0/code/html/custom.js#L540-L543

xoseperez, 2018-06-20:
I have pushed a fix for the multiple relays in the web UI issue. Relay numeration is zero-based everywhere except (for historical reasons) in the hardware.h file, where they are 1-based.

closed 2018-06-21 by xoseperez