xoseperez/espurna #957: espurna.ino.cpp.o - missing

icevoodoo, 2018-06-20:
Starting from today, i can't compile new firmware using PlatformIO. 
Please see the message bellow:

 **[.pioenvs\itead-sonoff-th\src\espurna.ino.cpp.o] The system cannot find the file specified**

Please some guidance to resolve this...

gn0st1c, 2018-06-20:
i just tried the latest dev branch with success;
`Environment itead-sonoff-th                     [SUCCESS]`

maybe running "Clean" task may help?

xoseperez, 2018-06-20:
I have recently introduced gulp builder to the main build process and it is triggered before `espurna.ino.cpp.o` is created (check the `extra_script.py` file). I can build it without a problem but Travis is also struggling there. Let me check what's going on...

mcspr, 2018-06-20:
I think gulp should be `node node_modules/gulp/bin/gulp.js`. pio eats exceptions with its own wrapper, so they are a bit burried

Also, there is a merge bug for pio.ini. extra_scritps variables are not closed properly. One of my installations of pio throws exceptions about it. Recent one ignores it somehow.
```
--- a/code/platformio.ini
+++ b/code/platformio.ini
@@ -246,7 +246,7 @@ upload_speed = 115200
 upload_port = ${common.upload_port}
 upload_flags = ${common.upload_flags}
 monitor_speed = 115200
-extra_scripts = ${common.extra_scripts
+extra_scripts = ${common.extra_scripts}
 modules = ${common.modules}

 # ------------------------------------------------------------------------------
@@ -276,7 +276,6 @@ upload_port = ${common.upload_port}
 upload_flags = ${common.upload_flags}
 monitor_speed = 115200
 extra_scripts = ${common.extra_scripts}
-extra_scripts = ${common.extra_scripts
 modules = ${common.modules} sensor

 [env:tinkerman-espurna-h08]
```

xoseperez, 2018-06-20:
Now it should be working fine (it is in Travis at least).

xoseperez, 2018-06-20:
Thanks @icevoodoo  and @mcspr

icevoodoo, 2018-06-20:
I have cloned again the project to be sure, and now the compile was OK but with one error in Scanning dependencies... Dependency Graph, and the check size of firmware.bin reportes as 0, see below:

**WEBUI_MODULES="" node node_modules/gulp/bin/gulp.js
'WEBUI_MODULES' is not recognized as an internal or external command,
operable program or batch file.
*** Error 1**

and this:

check_size([".pioenvs\itead-sonoff-th\firmware.bin"], [".pioenvs\itead-sonoff-th\firmware.elf"])
**Binary size: 0 bytes**
 [SUCCESS] Took 112.84 seconds

```
Processing itead-sonoff-th (framework: arduino; platform: espressif8266@1.5.0; board: esp01_1m)
 
Detected non-PlatformIO `modules` option in `[env:itead-sonoff-th]` section
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: Espressif 8266 > Espressif Generic ESP8266 ESP-01 1M
SYSTEM: ESP8266 80MHz 80KB RAM (1MB Flash)

Converting espurna.ino
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 47 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ArduinoJson> 5.13.2
|-- <AsyncMqttClient> 0.8.1 #ddbf4d1
|   |-- <ESPAsyncTCP> 1.1.3 #55cd520
|   |   |-- <ESP8266WiFi> 1.0
|-- <Brzo I2C> 1.3.2
|-- <DebounceEvent> 2.0.1 #e7dc564
|-- <EEPROM_Rotate> 0.9.1 #9c7d14f
|   |-- <EEPROM> 1.0
|-- <Embedis> 1.2.0
|-- <EspSoftwareSerial> 3.4.1 #5378868
|-- <ESPAsyncTCP> 1.1.3 #55cd520
|   |-- <ESP8266WiFi> 1.0
|-- <ESP Async WebServer> 1.1.0 #05306e4
|   |-- <ESPAsyncTCP> 1.1.3 #55cd520
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoJson> 5.13.2
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <FauxmoESP> 2.4.2 #50cbcf3
|   |-- <ESPAsyncTCP> 1.1.3 #55cd520
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <HLW8012> 1.1.0 #cc3baec
|-- <IRremoteESP8266> 2.2.0 #4d57a2b
|-- <JustWifi> 2.0.0 #fdf4804
|   |-- <ESP8266WiFi> 1.0
|-- <mDNSResolver> 0.1 #4cfcda1
|   |-- <ESP8266WiFi> 1.0
|-- <my92xx> 3.0.1 #cda124a
|-- <NoFUSS> 0.2.5 #1835206
|   |-- <ArduinoJson> 5.13.2
|   |-- <ESP8266httpUpdate> 1.1
|   |   |-- <ESP8266HTTPClient> 1.1
|   |   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|-- <NtpClientLib> 2.5.0 #0016a59
|   |-- <Time> 1.5 #ecb2bb1
|   |-- <ESP8266WiFi> 1.0
|   |-- <Ethernet(esp8266)> 1.0.4
|   |   |-- <SPI> 1.0
|-- <OneWire> 2.3.4
|-- <PZEM004T> 1.1.0
|   |-- <EspSoftwareSerial> 3.4.1 #5378868
|-- <PubSubClient> 2.6
|-- <rc-switch> 2.6.2
|-- <Time> 1.5 #ecb2bb1
|-- <ESP8266WiFi> 1.0
|-- <DNSServer> 1.1.0
|   |-- <ESP8266WiFi> 1.0
|-- <Wire> 1.0
|-- <ESP8266SSDP>
|   |-- <ESP8266WiFi> 1.0
|-- <Hash> 1.0
|-- <ArduinoOTA> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
|-- <Ticker> 1.0
WEBUI_MODULES="" node node_modules/gulp/bin/gulp.js
'WEBUI_MODULES' is not recognized as an internal or external command,
operable program or batch file.
*** Error 1

Compiling .pioenvs\itead-sonoff-th\src\espurna.ino.cpp.o
Compiling .pioenvs\itead-sonoff-th\src\libs\fs_math.c.o
Compiling .pioenvs\itead-sonoff-th\src\libs\pwm.c.o
Archiving .pioenvs\itead-sonoff-th\libbdc\libArduinoJson_ID64.a
....
.....
Linking .pioenvs\itead-sonoff-th\firmware.elf

Checking size .pioenvs\itead-sonoff-th\firmware.elf
Building .pioenvs\itead-sonoff-th\firmware.bin
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [=====     ]  54.0% (used 44240 bytes from 81920 bytes)
PROGRAM: [=====     ]  46.4% (used 486715 bytes from 1048576 bytes)
check_size([".pioenvs\itead-sonoff-th\firmware.bin"], [".pioenvs\itead-sonoff-th\firmware.elf"])
Binary size: 0 bytes
 [SUCCESS] Took 112.84 seconds
```

mcspr, 2018-06-20:
@icevoodoo Don't mind this ;) Working on a fix. Rebuild it - it will show correct size

xoseperez, 2018-06-21:
I have added code to check if `node` is in the path before trying to build the web UI. If it's not it will use the default image which has everything.

xoseperez, 2018-06-22:
Finally, 1.1.3.0 is out with a pre-build of all possible images so the reason for this issue has been completely removed. Closing

closed 2018-06-22 by xoseperez