xoseperez/espurna #965: Can't see bme280 sensor on status page in dev.

poulch74, 2018-06-21:
In master it works fine.

Support for bme and i2c enabled and in debug log sensors found, but only one relay is shown in status page..

000281] [INIT] BOARD: POULCH_SHAPEsp_RELAY
[000284] [INIT] SUPPORT: ALEXA BROKER DEBUG_SERIAL DEBUG_TELNET DEBUG_WEB DOMOTICZ HOMEASSISTANT I2C MDNS_SERVER MQTT NTP SCHEDULER SENSOR TELNET TERMINAL THINGSPEAK WEB 
[000299] [INIT] SENSORS: BMX280 

 [TELNET] Listening on port 23
[000336] [WEBSERVER] Webserver running on port 80
[000339] [RELAY] Retrieving mask: 0
[000339] [RELAY] Relay #0 boot mode 0
[000340] [RELAY] #0 set to OFF
[000341] [RELAY] Number of relays: 1
[000343] [BUTTON] Number of buttons: 0
[000347] [LED] Number of leds: 2
[000350] [MQTT] Async ENABLED, SSL DISABLED, Autoconnect ENABLED
[000363] [I2C] Using GPIO4 for SDA and GPIO5 for SCL
[000376] [I2C] Device found at address 0x68
[000378] [I2C] Device found at address 0x76
[000381] [THINGSPEAK] Async ENABLED, SSL DISABLED
[000383] [SENSOR] Initializing BMP280 @ I2C (0x00)
[000394] [I2C] Address 0x76 locked
[000408] [SENSOR]  -> temperature:0
[000409] [SENSOR]  -> pressure:0
[000409] [SENSOR]  -> humidity:0
[000431] [MAIN] Uptime: 0 seconds
[000431] [MAIN] Free heap: 18592 bytes
[000432] [MAIN] Power: 3401 mV

I am trying add my own custom board, so i testing everything and thinking about modifications...
at 0x68 address - ds3231, at 0x76 - bme280...

xoseperez, 2018-06-21:
I'm introducing a new feature in the dev branch to shrink the size of the web UI. It basically removed unneeded code unless you explicitly specify the module to load. At the moment these modules are disabled by default: sensor, rfbridge and light. 

I'm guessing you are building ESPurna for a custom device using PlatformIO. Is that true? Because that's the only way to build the web UI without does modules. If it's so, you might want to force adding the sensor module to the build using this setting in the environment (check other boards in the platformio.ini file):

```
module = sensor
```

poulch74, 2018-06-21:
I am using arduino ide (actually vs code with plugin)

xoseperez, 2018-06-21:
Can you check in the web code (Ctrl+U in the browser) if there is a `<div id="magnitudes">` tag?

poulch74, 2018-06-21:
in current prebuild DATA\index.html.gz no such tag...

xoseperez, 2018-06-21:
Can you update your local code (`git pull`) and retry?

poulch74, 2018-06-21:
i have updated code and just download current prebuild index.html.gz from repo... no such tag

poulch74, 2018-07-01:
I check it and i see bme280 sensor data :)

closed 2018-07-01 by poulch74