xoseperez/espurna #966: Generic JSON POST?

schweini, 2018-06-21:
I just got this awesome firmware to run on my Sonoff POW! Thanks!

I'm pulling data via the REST API via NAT on my router to a server right now. Works great.

But it would be way easier to tell the ESP8266 to periodically POST all its status data to a given URL.
Is this possible or on the horizon?

If not, does anyone have a pointer where I might try to insert this functionality into the code?

xoseperez, 2018-06-22:
This is not possible at the moment but a feasible option for the future. The idea would be to have an `http` module that would post the data from sensors (and relay statuses and so on) very much like the `influxdb` modules does.

schweini, 2018-06-22:
But is there already a ESPurna mechanism for executing something else (like
a plugin) every X minutes?

xoseperez, 2018-06-22:
You have to handle timers in your plugin. ESPurna is moving from an "everybody-knowns-it-all" paradigm to a more modular event-based one. You might notice different approaches in the code. You can study the `broker` module or the `influxdb` module for different approaches.