xoseperez/espurna #971: MHZ19 CO2 problem

MariusLaurinaitis, 2018-06-22:
Hello Xose,, I have same problem witth MHZ19 , maybe here's the same problem with the serial?

Thanks

[image]

[image]

MariusLaurinaitis, 2018-06-22:
UPDATE:
Espurna:
MHZ19 @ SwSerial(13,15) 
MHZ19_RX_PIN -> 13
MHZ19_TX_PIN -> 15

I did not compil the new one, just:
I swapped the wires only (I'll switch to RX and TX) and worked. But maybe you need to specify a PIN so that it does not mix others?

MHZ19 rx -> Wemos pin 15
MH19 TX -> Wemos pin 13

[image]

xoseperez, 2018-06-22:
Sorry, do you mean GPIO 13 must be connected to the MHZ19 TX?

MariusLaurinaitis, 2018-06-22:
Yes: GPIO 13 must be connected to the MHZ19 TX 
Thats work for me

xoseperez, 2018-06-23:
Well, that's right, in the definition 13 is RX (from the device point of view) and must be connected to sensor TX.

IvanUA, 2018-06-30:
Hi! I have the same problem - Error reading data from MH-Z19.
Power source for MH-Z19 - external 5V.

Is there any way to debug MHZ19?

MariusLaurinaitis, 2018-06-30:
did you swapped RX and TX?

IvanUA, 2018-06-30:
Of course, many times.
Gpio 13,15 (rx,tx and vice versa)
Gpio 1,3
No success

MH-Z19 old one (1st generation)

xoseperez, 2018-07-01:
I'm not much into these sensors and I don't know what changes were there between different generations... @IvanUA, do you have a working code for your sensor?

IvanUA, 2018-07-03:
Hi. Now I'm not sure about working condition of my MHZ19 (couldn't get it work directly through USB-UART). I'll will take a try with another one.  Thank you.

jsponz, 2018-07-20:
@MariusLaurinaitis, I have some problems when compiling for having support for MHZ19 -> https://github.com/xoseperez/espurna/issues/1063

Any idea? 

Thanks!

MariusLaurinaitis, 2018-07-20:
did you swapped RX and TX? 

P.S.
UPDATE:
Espurna:
MHZ19 @ SwSerial(13,15)
MHZ19_RX_PIN -> 13
MHZ19_TX_PIN -> 15

I did not compil the new one, just:
I swapped the wires only (I'll switch to RX and TX) and worked. But maybe you need to specify a PIN so that it does not mix others?

MHZ19 rx -> Wemos pin 15
MH19 TX -> Wemos pin 13

jsponz, 2018-07-20:
I tried it as well, but the same library error. I have a new installation, following all the steps of the guideline...

Which libraries are you using? 

Thanks!

MariusLaurinaitis, 2018-07-20:
I tried 1.13.1 with all libraries

MariusLaurinaitis, 2018-07-20:
I will try in the evening

derlucas, 2018-08-19:
try skipping the checksum check in MHZ19Sensor.h around line 181. 

`if ((buffer[0] == 0xFF) && (buffer[1] == 0x86)) {`

closed 2018-10-25 by stale[bot]

berezovskyi-oleksandr, 2018-11-01:
I believe that I faced same problem with checksum calculation.

In calculation loop, there is next line:
`for (unsigned char i = 1; i < MHZ19_REQUEST_LEN-1; i++) {`
This definition will let sum from 1 till 7 (MHZ19_REQUEST_LEN - 1). 

According to [datasheet (page 9)](https://www.winsen-sensor.com/d/files/PDF/Infrared%20Gas%20Sensor/NDIR%20CO2%20SENSOR/MH-Z19%20CO2%20Ver1.0.pdf) there should be sum from 1 till 8.

After replacing MHZ19_REQUEST_LEN with MHZ19_RESPONSE_LEN , calculations became valid.

ruimarinho, 2018-11-02:
@berezovskyi-oleksandr I have applied your suggested change but I still have the same issue. Tried multiple pins as well.