xoseperez/espurna #972: Cant compile 1.13.0

skyynet, 2018-06-22:
Every update I have problems with missing updated libs. I assume it's the same with this one.
After updating from 1.12.6 to 1.30.0 I now get this error while compiling
`In file included from sketch/config/all.h:32:0,
                 from /Users/ingo/espurna/code/espurna/espurna.ino:22:
sketch/config/prototypes.h:13:27: fatal error: EEPROM_Rotate.h: No such file or directory
 #include <EEPROM_Rotate.h>
                           ^
compilation terminated.
exit status 1
Fehler beim Kompilieren für das Board Generic ESP8266 `Module.``

I already upgraded justwifi to 2.0 as this lib was mentioned in the changelog.
How can I achieve every used lib to be updated automatically? I can see updates in the libs section of Arduino IDE but yours are always missing.

xoseperez, 2018-06-22:
I'm sorry I do not push my libraries to the Arduino library manager. I will investigate how to do it. In the meantime please install the EEPROM_Rotate library from the repo: https://github.com/xoseperez/eeprom_rotate.

skyynet, 2018-06-24:
Thanks for the info.

I also had to revert to ArduinoJson 5.x from the former installed 6.0.1-beta which worked for 1.12.6. Now it compiles.

If you can't push your libs to the Arduino Library Manager it would be enough to add a textfile with all used libs and their versions. By now every update gave me trouble just because of the (unknown) used libs.

closed 2018-06-24 by skyynet

xoseperez, 2018-06-24:
Yes, ArduinoJson 6 is not backwards compatible and not supported yet.

Actually, the platformio.ini file in the code folder defines all the
dependencies (versions and revisions).