xoseperez/espurna #975: Two compile errors V1.13.0

AlbertWeterings, 2018-06-23:
I have two issues when compiling ESPURNA v1.13.0 for two types of hardware in the Arduino IDE. Anyone have an idea how to solve this?

When compiling for ITEAD_SONOFF_RFBRIDGE

web:24: error: #elif with no expression
 #elif ITEAD_SONOFF_RFBRIDGE
exit status 1

Wen compiling for GIZWITS_WITTY_CLOUD
sketch\espurna.ino.cpp.o: In function `_fromHSV(char const*)':
------\code\espurna/utils.ino:436: undefined reference to `fs_pow'
------\code\espurna/utils.ino:436: undefined reference to `fs_log'
sketch\espurna.ino.cpp.o: In function `_setRGBInputValue':
------\code\espurna/utils.ino:436: undefined reference to `fs_pow'
sketch\espurna.ino.cpp.o: In function `_fromKelvin(unsigned long)':
------\code\espurna/utils.ino:436: undefined reference to `fs_pow'
------\code\espurna/utils.ino:436: undefined reference to `fs_log'
------\code\espurna/utils.ino:436: undefined reference to `fs_log'
sketch\espurna.ino.cpp.o: In function `_shadow()':
------\code\espurna/utils.ino:436: undefined reference to `pwm_set_duty'
------\code\espurna/utils.ino:436: undefined reference to `pwm_start'
sketch\espurna.ino.cpp.o: In function `_lightProviderUpdate()':
------\code\espurna/utils.ino:436: undefined reference to `pwm_set_duty'
------\code\espurna/utils.ino:436: undefined reference to `pwm_start'
sketch\espurna.ino.cpp.o: In function `~function':
------\code\espurna/utils.ino:436: undefined reference to `pwm_init'
sketch\espurna.ino.cpp.o: In function `lightSetup()':
------\code\espurna/utils.ino:436: undefined reference to `pwm_init'
------\code\espurna/utils.ino:436: undefined reference to `pwm_start'
collect2.exe: error: ld returned 1 exit status
exit status 1

xoseperez, 2018-06-23:
The first one should probably be:

```
#elif defined(ITEAD_SONOFF_RFBRIDGE)
```
The second one looks like it's not reading the `fs_math.*` files in the `lib` folder...

xoseperez, 2018-06-23:
Problem is with the Arduino IDE not parsing correctly files outside the main folder... it's not the first time it happens to me, I just seem to forget about it, sorry. I have pushed changes to `dev` to fix it.