xoseperez/espurna #976: Backup settings are not JSON-compliant

meyergru, 2018-06-23:
In reference to issue #896:

I just tried version 1.13.0 (the travis build), but a created backup file is still corrupted and cannot be imported again via the GUI.

My resulting backup file has 125 lines and is ~3000 bytes now (before with 1.12.6 it was ~1500 bytes) and each line looks complete, so it seems as if it is not truncated somewhere in the middle. It ends like so:

`
  "relayOnDisc5": "0",
  "relayOnDisc6": "0",
  "relayOnDisc7": "0",
  "adminPass": "XXXXX"
`

I think the problem is mainly because of a missing closing bracket which violates JSON syntax rules. Maybe that was forgotten in the new output scheme invented to fix #896?

BTW:

    I cannot re-open the old issue #896 myself.
    The release binaries for 1.13.0 are not online yet...

xoseperez, 2018-06-23:
I got it created correctly and restored correctly too:

```
{
  "app": "ESPURNA",
  "version": "1.13.0",
  "ssid0": "ssid",
  "pass0": "pass",
  "adminPass": "fibonacci",
  "hostname": "ESPURNA-B93C93",
  "boardName": "WEMOS_D1_MINI_RELAYSHIELD",
  "relayBoot0": "0",
  "relayPulse0": "0",
  "relayTime0": "0.00",
  "cfg": "3",
  "board": "3",
  "ledGPIO0": "2",
  "ledLogic0": "1",
  "btnGPIO0": "0",
  "btnRelay0": "0",
  "relayGPIO0": "5",
  "relayType0": "0",
  "mqttServer": "192.168.1.11",
  "mqttPort": "1883",
  "mqttEnabled": "1"
}
```

meyergru, 2018-06-23:
Sorry, my bad: The travis build I used was different from the now released binary version.

The official release 1.13.0 works. Thanks!

closed 2018-06-23 by meyergru