xoseperez/espurna #977: Freeze during cold start

kuppe234, 2018-06-23:
Hello,
I have a problem with cold start since v1.13.0. The update from v1.12.0 works fine, reboot runs through.
If itead S20 is disconnected from the power supply and restarted, it freezes. Only new flashing over the serial port can revive the outlet. But the problem remains.
What could be the reason.
In the attachment the output from the serial connection on cold start.
Greeting, kuppe.
[itead-s20_coldstart.bin.gz](https://github.com/xoseperez/espurna/files/2129884/itead-s20_coldstart.bin.gz)

kuppe234, 2018-06-23:
Additional Informations:
after flash over serial the Flash mode is DOUT,
after flash over Web the Flash mode is QIO

With QIO the coldstart fail.

kuppe234, 2018-06-23:
How can I set flash mode hard to DOUT for WEB-Update?

xoseperez, 2018-06-23:
The flash mode is configured on build time, you must be building it using QIO.

kuppe234, 2018-06-24:
Hello,
if I flash using QIO from serial I have the same problem. The S20 is freezing. The only mode that works is DOUT (also by Rfbridge and LED Controller  20).
How can I do that build time configuration? In platformio.ini is configured "board_build.flash_mode = dout".
Greetings, kuppe.

kuppe234, 2018-06-26:
Hello,
I had an issue with the platformio installation. After reinstall everything works.
I close this.
Greetings, kuppe.

closed 2018-06-26 by kuppe234