xoseperez/espurna #980: is the itead Sonoff S26 supported?

akosmaroy, 2018-06-23:
Hi,

I see a cheaper version of the S20 called the S26 on the Sonoff site: https://www.itead.cc/sonoff-s26-wifi-smart-plug.html

I wonder if this is also supported by ESPurna?

Akos

xoseperez, 2018-06-23:
It looks like it should work using the S20 image.

akosmaroy, 2018-08-02:
how did you flash the S20 image on it?

ruimarinho, 2018-08-02:
The usual process — `pio run -t upload -e itead-s20`.

xoseperez, 2018-08-04:
@ruimarinho I don't see a reason for the issues you are having. Are they only happening in the S26?

battika, 2018-09-11:
Hello,

Just got my Sonoff S26 and soldered all jumper cables like in the picture above provided by @ruimarinho - thanks for that. It is a very complicated effort compared to the S20.

I connected my USB-serial adaptor to the the 3V3, GND, ETX and ERX pins and saw the blue LED started flashing. Unfortunately after a while my adapter got hot and died on me. It seems this module drew too much power for it to be able to handle it. I had another USB-serial adaptor at hand but I decided to power the module externally using a breadboard Arduino power supply this time which turned out to be a good decision. 

I flashed the device using the following process:
1. Downloaded the latest ESPurna release for the Sonoff S20 from [here](https://github.com/xoseperez/espurna/releases/download/1.13.2/espurna-1.13.2-itead-s20.bin) 
2. Powered on the module while keeping the switch on the main board pressed. It is actually harder than it sounds as the switch on the main board is not very sensitive soI had to press it hard. When the device is in flash mode the blue LED should NOT be blinking. If it starts blinking you need to press the main board button harder and cycle the power supply.
3. I was using the esptool Python script for flashing that can be downloaded from [here](https://github.com/espressif/esptool). First you need to locate your USB-Serial adaptor port, it really depends on your OS and the type of it. I am using a Mac and my port turned out to be `/dev/cu.SLAB_USBtoUART`
4. Use the following command to flash the image using esptool but make sure you change the port and the filename to your needs: 
```
esptool.py --port /dev/cu.SLAB_USBtoUART write_flash -fs 1MB -fm dout 0x0 /Downloads/espurna-1.13.2-itead-s20.bin
```
5. Once done, cycle the power the blue LED on-board should start flashing and you can get connected to the ESPURNA-xxxxxx network and start the configuration. If you want to speed up the configuration steps and you are still connected to the device via the serial port, open your terminal program using baud 115200 and configure the wifi network using the following commands:
```
set ssid0 <YOUR_WIFI_SSID>
set pass0 <YOUR_WIFI_PASSWORD>
wifi.reset
```
You should see the S26 connecting to the configured wifi network:
```
[068730] [WIFI] Connecting to BSSID: AA:BB:CC:DD:EE:FF CH: 06, RSSI: -63, SEC: WPA2, SSID: <YOUR_WIFI_SSID>
```

I don't have any reboot occasions after changing the relay status or by configuring additional settings however I am only using this device for an hour. I have also connected it to my MQTT broker and it works fine.

xoseperez, 2018-09-11:
Cool, I have added info for the S26 to the Wiki, referencing this issue.
Thank you!

closed 2018-09-11 by xoseperez

battika, 2018-09-11:
Hello @xoseperez , thanks a lot for adding the instructions to the Wiki. After 6 hours it still runs fine without any hiccups. You might want to consider renaming the image to S20-S26 :)

mgaida11, 2018-10-08:
Thanks a lot for the description for the S26.

It was a real pain to solder the tiny strings to the tiny spots. Since I'm not the most professional solderer, the pad got burned from the PCB (oops) and so the whole device was on the way to the next waste bin. 

But then I decided to give flashing by air a chance (osmota). Since the firmware was uptodate ;-/ or whatever the whole thing led me to nowhere.

Never give up, I thought. So I started to fizzle with the tine PCB-hairy-tiny-things and managed to connect everything (checked): TX, RX, VCC and GND. But: the bootloader doesn't want to speak to me or to the esptool: 
"A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header".

The only thing I can do (as before that day) is to sonnect  using the Sonoff-App and switch on/off - BUT ONLY in the LAN. 

Does anyone here have an idea about this stange diagnosis? 

A cheerio the guys here doing such wonderful things! I love you.
Michael

battika, 2018-10-08:
When you power on the device, did you keep your switch on the main board pressed? It is important, without it esptool will not recognize your device nor you will be able to flash the firmware.

mgaida11, 2018-10-08:
Hi Battika, 

yes I did - and I pressed it (as described above) quite a lot ;-)

The light-play is also strange. 

When the device is pluged into 230V and the button is pressed, it has different states, but it blinks in different frequensies.

When I turn it on (without 230V) but with the help of a stable 3,3V, and I push the key when connected, the light goes off. Like described. 

So I think I'm doing what the recipe says.

giuaig, 2019-01-19:
I see io2 on the front and io4/io5 on the back, can they be safely used to connect sensors?

I'm asking this because in the POWr2 the ground "is connected to neutral mains, so it's not safe to do that" as stated in https://github.com/xoseperez/espurna/issues/1463#issuecomment-451495927 and so i'm a bit scared to try :P

sfromis, 2019-03-05:
> I see io2 on the front and io4/io5 on the back, can they be safely used to connect sensors?

You're right to be scared, but more than just a "bit" :-)

Generally, this is much too unsafe with mains power input, as anything connected will have a risk of getting mains power through the required ground pin. It is very common to be able to plug mains plugs into the socket either way, meaning that in 50% of cases you could be feeding mains power to the sensor, button or whatever, not designed to protect against such voltages and power levels. Potential outcomes could include fried chips, electrical shocks, fire or death.

I'm surprised that such a risky design went into production, and even with very tempting marked header holes. It would be safe to use without mains input, but then there's not much point in using a device specifically intended for switching mains power.

CarlosGS, 2020-05-03:
For anyone reaching this thread - also try swapping the TX/RX pins... T_T

fips21, 2022-10-21:
In the meantime the S26 has another layout and a ESP8285. Now the pins are on the main board and not on the small ecu board any more. Unfortunately it's not stable for me. It reboots round about every minute. Tried different versions of espurna without luck. :-(

mcspr, 2022-10-22:
> In the meantime the S26 has another layout and a ESP8285. Now the pins are on the main board and not on the small ecu board any more. Unfortunately it's not stable for me. It reboots round about every minute. Tried different versions of espurna without luck. :-(

Different versions of espurna could mean many things :) If device reboots, this usually means some kind of crash that you should be able to recover in the DEBUG panel of webui or through telnet connection by sending `crash` command. If you are using snapshot builds, should be pretty easy to track where the issue comes from. Please open a new issue, though, at least mentioning which versions you used