xoseperez/espurna #981: GIZWITS_WITTY_CLOUD sensord do not appear

AlbertWeterings, 2018-06-23:
I have a few GIZWITS_WITTY_CLOUD devices build firmware in Arduino IDE.
Switch #0 and the channels appear in the web interface but no sign of the analog sensor and some Dallas sensors I have connected to it. Also the sensor menu item is missing. The values of the sensors are available in the MQTT messages

AlbertWeterings, 2018-06-23:
Found where the issue is coming from but don't have a solution. This occurs due to "static/index.light.html.gz.h" is used in web.ino if I uncomment "static/index.light.html.gz.h" and use "static/index.sensor.html.gz.h" in the web interface shows the sensors but Lights it then gone.

It seems not possible to have both.

xoseperez, 2018-06-23:
I see. The thing is that the Witty is a light (so LIGHT_PROVIDER != LIGHT_PROVIDER_NONE). If you add sensors to it if will still use the `static/index.light.html.gz.h`. The solution is to add a `static/index.all.html.gz.h` image and load it when this happens. I'd like to create individual images for each device but that would add too much build time at the moment and increase the build complexity. It's something to reconsider in the future.

AlbertWeterings, 2018-06-23:
Initially it is a RGB light with a analog sensor. Maybe it is a idea to be able to configure it as it is now, and have a option to configure it as a device with sensors and just 3 LEDs with a on off switch.

I use it with a Dallas sensors and according to the readout of the sensor I have the LED light red, green or blue.

AlbertWeterings, 2018-06-23:
Disabling in hardware.h
//    #define RELAY_PROVIDER      RELAY_PROVIDER_LIGHT
//    #define LIGHT_PROVIDER      LIGHT_PROVIDER_DIMMER

Brings up the sensors. I will have to check if it is still possible now to control the RGB LED via MQTT if that works in my application this is a good solution.

xoseperez, 2018-06-23:
I have the fix, I'm checking if other boards have been affected too..

xoseperez, 2018-06-23:
@AlbertWeterings Can you test the code in the `dev` branch?

AlbertWeterings, 2018-06-23:
Just flashed it in one device and it now shows both light and sensors. In the morning I will flash it in all devices that are using multiple modules.

AlbertWeterings, 2018-06-24:
Ok during compiling for RFBRIDGE I run into a error.
web:44: error: operator '==' has no left operand
 #if ITEAD_SONOFF_RFBRIDGE == 1
exit status 1
operator '==' has no left operand

Changing line 44 to "#if ITEAD_SONOFF_RFBRIDGE" is solving this issue

closed 2018-07-10 by xoseperez