xoseperez/espurna #984: Sonoff RF Bridge 433 - Soft WDT reset

gernst48, 2018-06-24:
Hi,

I get the following issue with one of my two boards.
This happens after several power up cyles.
The other board is working perfect.
If I flash the board again, it is working but again after several power up
cycles the same happens again and again.
What could I do ?

Ernst

-----------------------------------------------------------------------------

[000183] [INIT] ESPURNA 1.12.6
[000183] [INIT] xose.perez@gmail.com
[000183] [INIT] http://tinkerman.cat

[000184] [INIT] CPU chip ID: 0xCAD98E
[000185] [INIT] CPU frequency: 80 MHz
[000189] [INIT] SDK version: 1.5.3(aec24ac9)
[000193] [INIT] Core version: 2.3.0
[000196] [INIT] Core revision:
[000198]
[000200] [INIT] Flash chip ID: 0x144051
[000203] [INIT] Flash speed: 40000000 Hz
[000206] [INIT] Flash mode: DOUT
[000209]
[000210] [INIT] Flash sector size:     4096 bytes
[000215] [INIT] Flash size (CHIP):  1048576 bytes
[000219] [INIT] Flash size (SDK):   1048576 bytes /  256 sectors
[000225] [INIT] Firmware size:       476688 bytes /  117 sectors
[000230] [INIT] OTA size:            548864 bytes /  134 sectors
[000236] [INIT] EEPROM size:           4096 bytes /    1 sectors
[000241] [INIT] Empty space:          16384 bytes /   4 sectors
[000247]
[000249] [INIT] BOARD: ITEAD_SONOFF_RFBRIDGE
[000252] [INIT] SUPPORT: ALEXA BROKER DEBUG_SERIAL DEBUG_TELNET DEBUG_WEB DOMOTI
CZ HOMEASSISTANT MDNS_SERVER MQTT NTP SCHEDULER TELNET TERMINAL THINGSPEAK WEB

[000266] [INIT] Last reset reason: Power on
[000269] [INIT] Settings size: 2213 bytes
[000274] [INIT] Free heap: 29760 bytes
[000277] [INIT] Power: 3172 mV
[000279] [INIT] Power saving delay value: 10 ms
[000283]
[000308] [TELNET] Listening on port 23
[000310] [WEBSERVER] Webserver running on port 80
[000319] [RELAY] Retrieving mask: 0
[000320] [RELAY] Relay #0 boot mode 0
[000320] [RELAY] Relay #1 boot mode 0
[000320] [RELAY] Relay #2 boot mode 0
[000322] [RELAY] Relay #3 boot mode 0
[000325] [RELAY] Relay #4 boot mode 0
[000328] [RELAY] Relay #5 boot mode 0
[000331] [RELAY] Relay #6 boot mode 0
[000335] [RELAY] Relay #7 boot mode 0
[000338] [RELAY] #0 set to OFF
[000342] [RFBRIDGE] Enqueuing MESSAGE 'C00100001800140014' 4 time(s)
[000347] [RELAY] #1 set to OFF
[000351] [RFBRIDGE] Enqueuing MESSAGE 'C00100001800144014' 4 time(s)
[000355] [RELAY] #2 set to OFF
[000359] [RFBRIDGE] Enqueuing MESSAGE 'C00100001800141014' 4 time(s)
[000364] [RELAY] #3 set to OFF
[000368] [RFBRIDGE] Enqueuing MESSAGE 'C00100001800145014' 4 time(s)
[000373] [RELAY] #4 set to OFF
[000376] [RELAY] #5 set to OFF
[000379] [RFBRIDGE] Enqueuing MESSAGE 'C002000020818A9D00' 1 time(s)
[000384] [RELAY] #6 set to OFF
[000388] [RFBRIDGE] Enqueuing MESSAGE 'C002000020AE8A9D00' 1 time(s)
[000393] [RELAY] #7 set to OFF
[... 487 lines cut ...]
[000250] [INIT] SUPPORT: ALEXA BROKER DEBUG_SERIAL DEBUG_TELNET DEBUG_WEB DOMOTI
CZ HOMEASSISTANT MDNS_SERVER MQTT NTP SCHEDULER TELNET TERMINAL THINGSPEAK WEB

[000264] [INIT] Last reset reason: Software Watchdog
[000269] [INIT] Settings size: 2213 bytes
[000273] [INIT] Free heap: 30992 bytes
[000276] [INIT] Power: 3170 mV
[000278] [INIT] Power saving delay value: 10 ms
[000282]
[INIT] Device is in SAFE MODE
[000286]
[000291] [TELNET] Listening on port 23
[000292] [MAIN] Uptime: 0 seconds
[000293] [MAIN] Free heap: 29912 bytes
[000297] [MAIN] Power: 3170 mV
[000422] [WIFI] Creating access point
[000612] [WIFI] Captive portal enabled
[000612] [WIFI] MODE AP --------------------------------------
[000612] [WIFI] SSID  EGSONOFF2
[000613] [WIFI] PASS  pipsi1911
[000615] [WIFI] IP    192.168.4.1
[000618] [WIFI] MAC   DE:4F:22:CA:D9:8E
[000622] [WIFI] ----------------------------------------------

-----------------------------------------------------------------------------

wildwiz, 2018-06-24:
It might be worthy to install the new 1.13.0 firmware. It has onboard a more robust method to write the settings in the non volatile memory and your description makes me think of corrupted configuration data, whatever the cause.

xoseperez, 2018-06-29:
It looks like the device crashes when scanning networks, doesn't it? Can you type `wifi.scan` when the device is in SAFE MODE to check this?