xoseperez/espurna #985: Wireless router connection delay

analistamarcioprado, 2018-06-24:
Good evening.

I have one small problem:

I have 4 Sonoff Pow connected to a Wireless network. When the power runs out, all the equipment obviously turns off.

When the power comes back, the 4 Sonoff Pows connect faster than the wireless router.

As a router is not yet ready, Sonoff does not find the wireless SSID and each works as if it were an AP. In this way I lose the connection with them.

Is there any way to slow the Sonoff ligament? Or at least the connection to the AP?

Thank you.

xoseperez, 2018-06-25:
As @AlbertWeterings the devices should retry connecting after 3 minutes.

analistamarcioprado, 2018-06-25:
Ah yes, @xoseperez . In the tests I did I did not wait for the 3 minutes.

Thank you @AlbertWeterings .

This afternoon I'm going to test.

analistamarcioprado, 2018-06-25:
Analyzing the posts again @xoseperez  I realized that I did not understand well. Is this functionality already implemented?

closed 2018-06-25 by analistamarcioprado

frankforpresident, 2018-06-25:
@xoseperez, Any plans in making this setting configurable? It would be nice if we could set this retry to 1 minute or what every you like.

analistamarcioprado, 2018-06-25:
It would be perfect if we could set this time.